newaetech/chipwhisperer #248: Source code bundle for v5.1.3 missing submodule contents

dev-zzo, 2019-12-11:
The .tar.gz source code bundle version 5.1.3 available under releases does not include the contents of any of the submodules in the repo: ./jupyter ./openadc ./tutorials are all empty.

This breaks the installation flow as described in the official documentation.

dev-zzo, 2019-12-11:
I know the release notes say the jupyter thing is not included. It doesn't say what I need to do to get these installed properly to get started with CW.

Will it be okay to clone these manually from git? Will they work properly? How do I install them so they are available?

colinoflynn, 2019-12-11:
This is a bit "release issue" to be honest... we had previously packed those back into the the zip-file I think, but there is no way to tell github not to create it's automatically packaged version (which is missing the submodules). The latest release doesn't have that fixed one at all however.

Grabbing the missing folders from git will 100% be the recommended way. As a note I think you actually only need the "jupyter" folder, the "tutorials" is just pre-built examples, and "openadc" was things for building FPGA/hw.

m4xw, 2019-12-11:
I could recommend https://github.com/ingydotnet/git-subrepo

alex-dewar, 2022-01-27:
Closing as there's been no activity for a while

closed 2022-01-27 by alex-dewar