newaetech/chipwhisperer #250: SAM_CS2 Not connected

naasik12, 2019-12-30:
Hi I was reviewing the Schematics. Is there missing pins or connections from the schematics?

Kind Regards

colinoflynn, 2019-12-30:
Not sure what you mean? Is this for CW-Lite? There hasn't been updates in a long time, those schematics are "golden" in that PCB etc is done for that.

naasik12, 2019-12-30:
Hi Colin
SAM_CS2 has no target point.
Thanks for replying.

[image]

naasik12, 2019-12-30:
[image]

colinoflynn, 2019-12-30:
Ah gotcha! Yeah that goes nowhere, I think it was a possible usage, but it's a N/C in the end.

Anything with a connection will tell you the sheet & location on the sheet (i.e., SAM_CS1/7.5H means Sheet 7, Location 5-H). Although I think in my case I didn't actually use the schematic sheet that had XY marking, so they are useless (the marking of location is done automatically)

closed 2019-12-30 by colinoflynn