newaetech/chipwhisperer #251: Can't set trigger timeout in v5.1.1

dev-zzo, 2020-01-06:
For some reason, setting trigger timeout does not work in version 5.1.1.

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-3c028533c1db> in <module>
     29     scope.clock.clkgen_freq = (104e6) / 4
     30     scope.clock.adc_src = "clkgen_x4"
---> 31     scope.trigger.timeout = 3.0
     32     scope.trigger.triggers = "tio4"
     33     scope.io.tio1 = "serial_rx"

~/chipwhisperer/software/chipwhisperer/common/utils/util.py in __setattr__(self, name, value)
    363     def __setattr__(self, name, value):
    364         if hasattr(self, '_new_attributes_disabled') and self._new_attributes_disabled and not hasattr(self, name):  # would this create a new attribute?
--> 365             raise AttributeError("Attempt to set unknown attribute in %s"%self.__class__, name)
    366         super(DisableNewAttr, self).__setattr__(name, value)
    367 

AttributeError: ("Attempt to set unknown attribute in <class 'chipwhisperer.capture.scopes.cwhardware.ChipWhispererExtra.TriggerSettings'>", 'timeout')
```

Setting `triggers` works just fine.

dev-zzo, 2020-01-06:
I'm using CWLite if that's important (however docs do not mention this being hardware dependent).

dev-zzo, 2020-01-06:
Okay, I figured out this to be a documentation issue.

The docs at https://chipwhisperer.readthedocs.io/en/latest/api.html#chipwhisperer.scopes.OpenADC.adc note each member to be under `scope.trigger` while it should say `scope.adc` instead...

alex-dewar, 2020-02-10:
Sorry for the delay on this. It should be all fixed up in the latest dev commit and on readthedocs.

closed 2020-02-10 by alex-dewar