newaetech/chipwhisperer #256: Unable to program the target on a ChipWhisperer-Lite ARM

sopmacF, 2020-02-27:
I'm somehow no longer able to program the target.
I always get the following error message when trying to flash the target. 

**cw.program_target(scope, prog, fw_path)**

_Detected known STMF32: STM32F302xB(C)/303xB(C)

---------------------------------------------------------------------------
CmdException                              Traceback (most recent call last)
<ipython-input-17-b876eeece5c0> in <module>
----> 1 cw.program_target(scope, prog, fw_path)

~/work/projects/chipwhisperer/software/chipwhisperer/__init__.py in program_target(scope, prog_type, fw_path, **kwargs)
     52     prog.open()
     53     prog.find()
---> 54     prog.erase()
     55     prog.program(fw_path, memtype="flash", verify=True)
     56     prog.close()

~/work/projects/chipwhisperer/software/chipwhisperer/capture/api/programmers.py in func_wrapper(self, *args, **kwargs)
     68             self.scope.io.nrst = 'high_z'
     69         try:
---> 70             val = func(self, *args, **kwargs)
     71         finally:
     72             logging.debug('Restoring pdic, pdid, and nrst pin configuration')

~/work/projects/chipwhisperer/software/chipwhisperer/capture/api/programmers.py in erase(self)
    322         self.log("Erasing Chip")
    323         stm32f = self.stm32prog()
--> 324         stm32f.cmdEraseMemory()
    325 
    326     @save_and_restore_pins

~/work/projects/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in cmdEraseMemory(self, sectors)
    425     def cmdEraseMemory(self, sectors=None):
    426         if self.extended_erase:
--> 427             return self.cmdExtendedEraseMemory()
    428 
    429         if self.cmdGeneric(0x43):

~/work/projects/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in cmdExtendedEraseMemory(self)
    447 
    448     def cmdExtendedEraseMemory(self):
--> 449         if self.cmdGeneric(0x44):
    450             logging.debug("*** Extended Erase memory command")
    451             # Global mass erase

~/work/projects/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in cmdGeneric(self, cmd)
    329         self.sp.write(chr(cmd))
    330         self.sp.write(chr(cmd ^ 0xFF))  # Control byte
--> 331         return self._wait_for_ask(hex(cmd))
    332 
    333     def cmdGet(self):

~/work/projects/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    295                 if ask == 0x1F:
    296                     # NACK
--> 297                     raise CmdException("NACK " + info)
    298                 else:
    299                     # Unknown response

CmdException: NACK 0x44_

alex-dewar, 2020-02-27:
Hi,

Can you try running program_target(..., baud=38400)? If that doesn't work, try a few different values.

Alex

sopmacF, 2020-02-27:
Hi Alex,

Thx for the quick reply!
I tried the following baud rates (9600, 19200, 38400, 57600) unfortunately without success. Somehow strange, until yesterday everything was working perfectly. In the meanwhile, I installed the newest version (5.1.3). 

> **cw.program_target(scope, prog, fw_path, baud=9600)**
> 
> Serial baud rate = 9600
> Detected known STMF32: STM32F302xB(C)/303xB(C)
> 
> ---------------------------------------------------------------------------
> CmdException ...

Thank you again.

Cheers,
Fabio

sopmacF, 2020-02-28:
I managed to flash the board via openocd/gdb over ST-Link.

used openocd flags: _-c init -c "reset halt" -c "flash probe 0" -c "stm32f3x unlock 0" -c "reset init" -c "flash protect 0 0 last off" -c "stm32f3x mass_erase 0"_

```
@gdb: 
(gdb) target remote localhost:50000
Remote debugging using localhost:50000
0xfffffffe in ?? ()
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x3960 lma 0x8000190
Loading section .rodata, size 0x710 lma 0x8003af0
Loading section .ARM, size 0x8 lma 0x8004200
Loading section .init_array, size 0x4 lma 0x8004208
Loading section .fini_array, size 0x4 lma 0x800420c
Loading section .data, size 0x64 lma 0x8004210
Start address 0x80029f8, load size 17004
Transfer rate: 17 KB/sec, 2429 bytes/write.
(gdb)
```

Now I'm getting the same message as described in the issue #229.

**cw.program_target(scope, prog, fw_path, baud=38400)**

```
Serial baud rate = 38400
Detected known STMF32: STM32F302xB(C)/303xB(C)
Extended erase (0x44), this can take ten seconds or more
Attempting to program 4691 bytes at 0x8000000
STM32F Programming flash...
Serial baud rate = 38400

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
~/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    304         # wait for ask
--> 305         try:
    306             ask = self.sp.read(1)[0]

IndexError: list index out of range

During handling of the above exception, another exception occurred:

CmdException                              Traceback (most recent call last)
~/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in writeMemory(self, addr, data, smallblocks)
    644 
--> 645             try:
    646                 self.cmdWriteMemory(addr, data[offs:offs + block_size])

~/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in cmdWriteMemory(self, addr, data)
    439                     self.delay_func(5)
--> 440             self.sp.write(chr(crc))
    441             self._wait_for_ask("0x31 programming failed")

~/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    306             ask = self.sp.read(1)[0]
--> 307         except:
    308             raise CmdException("Can't read port or timeout (%s). Target didn't respond when an ack was expected." % traceback.format_exc())

CmdException: Can't read port or timeout (Traceback (most recent call last):
  File "/home/sopmac/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 305, in _wait_for_ask
    try:
IndexError: list index out of range
). Target didn't respond when an ack was expected.

During handling of the above exception, another exception occurred:

IndexError                                Traceback (most recent call last)
[... 88 lines cut ...]
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sopmac/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 645, in writeMemory
    try:
  File "/home/sopmac/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 440, in cmdWriteMemory
    self.sp.write(chr(crc))
  File "/home/sopmac/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 307, in _wait_for_ask
    except:
chipwhisperer.hardware.naeusb.programmer_stm32fserial.CmdException: Can't read port or timeout (Traceback (most recent call last):
  File "/home/sopmac/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 305, in _wait_for_ask
    try:
IndexError: list index out of range
). Target didn't respond when an ack was expected.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sopmac/opt/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 305, in _wait_for_ask
    try:
IndexError: list index out of range
). Target didn't respond when an ack was expected.
```

Any suggestions?

sopmacF, 2020-02-28:
After trying many approaches everything seems to work now.
My final solution: Flash the board once via OpenOCD/gdb (as described above) and then downgrade to version 5.1.1.

alex-dewar, 2020-03-05:
Hi, a quick update on this:

I think this was caused by a timeout being changed for the STM32 chips. I've reverted the change for non F4 boards. If possible, can you see if this fixes your issue?

Alex

colinoflynn, 2020-03-05:
Did we also ever figure out if the delay in serial.py (https://github.com/newaetech/chipwhisperer/commit/0f263f1a8033677c4a047ee0ef88d55bc34c047d#diff-b1e00e5048babd6ae5675cf348e51577) was useful to revert in some cases?

alex-dewar, 2020-03-05:
The only difference IIRC is that programming at lower baud rates is more likely to work due to the ChipWhisperer not reporting that the send buffer is full. I haven't found it to cause any other issues.

closed 2020-08-27 by alex-dewar