newaetech/chipwhisperer #258: Forums Forbidden?

colinoflynn, 2020-03-06:
Have had a few reports of getting 403 forbidden on forum.newe.com. Unclear the source here - let me know if you hit this.

adelapie, 2020-04-26:
Hello, I usually have this problem when I connect to forum.newe.com from France.

he-so, 2020-04-30:
Same here:
[forum.newae.com.har.txt](https://github.com/newaetech/chipwhisperer/files/4559823/forum.newae.com.har.txt)

  1   245 ms     1 ms     1 ms  fritz.box [2001:16b8:1c0b:4b00:3681:c4ff:fe36:2d77]
  2   138 ms     6 ms    10 ms  2001:1438::62:214:63:150
  3    82 ms     5 ms     5 ms  2001:1438:0:1::9:3a1
  4    14 ms    13 ms    13 ms  2001:1438:0:1::7:2302
  5     *        *        *     Zeitüberschreitung der Anforderung.
  6     *        *        *     Zeitüberschreitung der Anforderung.
  7     *        *       32 ms  be2182.ccr21.lpl01.atlas.cogentco.com [2001:550:0:1000::9a36:4df6]
  8     *        *      101 ms  be3042.ccr21.ymq01.atlas.cogentco.com [2001:550:0:1000::9a36:2ca2]
  9   101 ms   101 ms   101 ms  be2089.rcr21.ymq02.atlas.cogentco.com [2001:550:0:1000::9a36:2d72]
 10    97 ms    96 ms    96 ms  2001:550:2:6::12:2
 11    98 ms    98 ms    98 ms  2607:f748:0:f012::2
 12    97 ms    97 ms   120 ms  te6-4.v0700.cl-core05.ip6.mtl.iweb.com [2607:f748:0:1:0:5:6:1]
 13    97 ms    96 ms    96 ms  s233.panelboxmanager.com [2607:f748:1301:0:184:107:112:61]

colinoflynn, 2020-04-30:
Thanks for more info on this! Back on tracking it down... had only tried a few VPN end-points, didn't realize some made it more frequent. Might just need to swap the server around

colinoflynn, 2020-04-30:
Thanks for the traceroute - this might have been the jolt I needed. That IPv6 address is pointing to an old server - it's still used for some stuff so it's still in the DNS, but I _didn't notice_ it was still being resolved for the IPv6 version of forum.newae.com. This would explain why I never saw much of anything in the logs on the discourse server!

Updated now - let me know if you still get the issue in > 4 hours (DNS TTL should expire by then).

Right now it's just IPv4, shoudl always be going to 159.89.119.148 (I had temporarily redirected to another location, so things may be screwed for next few hours as everything propagates).

he-so, 2020-04-30:
Looks good, now👍

colinoflynn, 2020-04-30:
Thanks for confirming! IPv6 strikes again

closed 2020-04-30 by colinoflynn