newaetech/chipwhisperer #259: Scope timed out errors and no longer get any target output

tonkapango, 2020-03-23:
Unable to get any output from target.  Seems trigger is not firing and just get timeouts. e.g. "WARNING:root:Timeout in OpenADC capture(), trigger FORCED".   Note i have never had this problem until this month when reinstalling my setup at home.

background:
on ubuntu 19.04 device, with virtualbox 6.0.14, i used the Virtual_Machine.ChipWhisperer.Jupyter-5-1-1.zip VM image.  and i ran into the same exact issues as sopmacF #256, "unable to program the target on a CHW-Lite ARM". with both STM32F071 and SMT32F4 targets.  following that thread, just cloned chipwhisperer with alex's STM update, and now i can program the targets fine, but looks like trigger is not firing as scope.capture times out with no data from the target.

repro:
1. install latest virtualbox
2. using the 5-1-1 CW image, upgrade the the chipwhisperer code to (9cc598c855bd126de1908e22a60b8d7bcfb15810 fix).
3. Using the STM32F415 target
4. using Fault_1-Introduction_to_Clock_Glitch_Attacks.ipynb, execute the blocks to and including:

target.flush()
scope.arm()
reset_target(scope)

ret = scope.capture()
if ret:
    print("Scope capture timed out")
response = target.read(timeout = 10)
print(response)

5. observe we never see the output from the simple-glitch.c firmware "hello".

WARNING:root:Timeout in OpenADC capture(), trigger FORCED
WARNING:root:Timeout in OpenADC capture(), trigger FORCED

Scope capture timed out

alex-dewar, 2020-03-23:
Hi,

I'd guess that the issue is somewhere on the target side. To confirm, have you built the firmware with `PLATFORM=CW308_STM32F4`? If you have, please open `chipwhisperer/hardware/victims/firmware/hal/stm32f4/Makefile.stm32f4` and change the following lines from:

```
CFLAGS += -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fmessage-length=0 -ffunction-sections
CPPFLAGS += -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fmessage-length=0 -ffunction-sections
ASFLAGS += -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fmessage-length=0 -ffunction-sections
```
to:

```
CFLAGS += -mthumb -mfloat-abi=soft -fmessage-length=0 -ffunction-sections
CPPFLAGS += -mthumb -mfloat-abi=soft -fmessage-length=0 -ffunction-sections
ASFLAGS += -mthumb -mfloat-abi=soft -fmessage-length=0 -ffunction-sections
```
and rerun the tutorial?

Thanks,

Alex

tonkapango, 2020-03-23:
i was toggling between PLATFORM = 'CWLITEARM' and PLATFORM = 'CW308_STM32F4'. i opted to spend most of the time on CWLITEARM because i was seeing redefinition warnings with CW308_STM32F4.  But i just ran it again today with platform = CW308_STM32F4, and it worked! 

i was able to receive target output.   i did not have to modify the stm32f4 makefile.  the only introduced condition between now and then, was i had a segger jlink attached to the UFO baseboard.  So not sure how this issue was caused or why it is not manifested now.  But if it comes up again, i can try to modify the above flags.

thanks alex.

closed 2020-03-23 by tonkapango

colinoflynn, 2020-03-23:
Glad it worked out! FYI I've seen an issue where some of the pins with pull-ups float the wrong way, or possibly just aren't pulled to their state fast enough. Sometimes it's helpful to strap the pin PDIC pin low & reset the target, the following chunk of code does that:

    self.scope.io.pdic = False
    self.scope.io.nrst = False
    time.sleep(0.05)
    self.scope.io.nrst = True

I assume what has happened is the device didn't come out of reset when the pdic pin was pulled low correctly. It should be fixable in the SW, but try the above if it happens again.

You can also set nrst to None which floats the nrst pin. This lets you use the reset button the CW308 board too. Also if using a segger you'll need to set the nrst pin to floating! You can do for example the following:

    self.scope.io.pdic = False
    self.scope.io.nrst = False
    time.sleep(0.05)
    self.scope.io.nrst = True
    time.sleep(0.05)
    self.scope.io.nrst = None

tonkapango, 2020-03-25:
this is great. i have added the above snippet to reset target; and no issue receiving target output.  thanks for the tip on floating nrst when using segger jlink! (off issue topic, but if there is any additional guidance when using segger with CHW-lite, would be super helpful).