newaetech/chipwhisperer #260: trying to narrow down jtag connection issues.

tonkapango, 2020-03-27:
apologies for a general question. but i have the CHW FPGA connected to the CW308-04 which has an rv target. Also connected to the CW308-04, via JTAG, is a segger j-link.  with jlinkExe and openocd, i get "device not found" errors.  from my host machine i can see successfully connect to the jlink.

i am just wondering if there is anything on the CW308-04 that i should do to properly interface with a UFO target via CW308 JTAG (jumper settings, etc)?  just want to rule out anything physical, as i focus on device and target definitions in jlink and openocd.  assuming using programmers is a common use case, but didn't readily see specific setup info. thanks!

alex-dewar, 2020-03-27:
Hi,

There shouldn't be any special jumper settings needed on the CW308. One thing to make sure of is that the connector is on the right way around. Row 2 is the one closer to the target. 

[image]

It might be a little hard to see, but the key in that picture is pointing towards the STLink.

Alex

tonkapango, 2020-03-28:
thanks alex for confirming, yes i have the connecter plugged in correctly.  as i continue to dig down on the connection issue. it appears the J-Link cannot reset the target, maybe perhaps because the CW308 is controlling the clock?  will keep futzing but this command seems to work fine on a setups without cw308 (jlink and target only).

flashing command from freedom Studio
$ make upload
SEGGER J-Link Commander V6.70 (Compiled Mar 27 2020 14:37:00)
DLL version V6.70, compiled Mar 27 2020 14:36:47

Connecting to J-Link via USB...O.K.
Firmware: J-Link V11 compiled Mar 19 2020 11:11:08
Hardware version: V11.00
S/N: 601001907
License(s): RDI, FlashBP, FlashDL, JFlash, GDB
VTref=3.303V
Device "FE310" selected.

Connecting to target via JTAG
ConfigTargetSettings() start
ConfigTargetSettings() end
TotalIRLen = 5, IRPrint = 0x01
JTAG chain detection found 1 devices:
 #0 Id: 0x20000913, IRLen: 05, Unknown device
Debug architecture:
  RISC-V debug: 0.13
  AddrBits: 7
  DataBits: 32
  IdleClks: 5
Memory access:
  Via system bus: No
  Via ProgBuf: No (0 ProgBuf entries)
DataBuf: 0 entries
  autoexec[0] implemented: No
Failed to temporarily halt core during identification
Attaching to core failed. Trying again with connect under reset (ndmreset-bit)
Memory access:
Detected: RV32 core
CSR access via abs. commands: Yes

**************************
WARNING: CPU could not be halted
**************************

ConfigTargetSettings() start
ConfigTargetSettings() end
ConfigTargetSettings() start
ConfigTargetSettings() end

****** Error: Timeout while waiting for core to halt after reset and halt request
Communication timed out: Requested 4 bytes, received 0 bytes !
Specific core setup failed.
Could not read hardware status!
Cannot connect to target.
J-Link>Target connection not established yet but required for command.
Device "FE310" selected.

Selected interface (JTAG) is not supported by the connected probe.
J-Link>Target connection not established yet but required for command.
Device "FE310" selected.

Connecting to target via JTAG
Cannot connect to target.

alex-dewar, 2020-03-28:
Can you try setting `scope.io.nrst` to `"high_z"`?

tonkapango, 2020-03-29:
you are right. user error.  colin mentioned this on another thread, and while i had the below snippet in my notebook, i had actually executed another block which had nrst high when trying j-link connection. this has been rectified.  thanks alex.

SCOPETYPE = 'OPENADC'
PLATFORM = 'CWLITEARM'

%run "Helper_Scripts/Setup_Generic.ipynb"

scope.io.pdic = False
scope.io.nrst = False
time.sleep(0.05)
scope.io.nrst = True
time.sleep(0.05)
scope.io.nrst = None
print(scope)

Serial baud rate = 38400
INFO: Found ChipWhisperer😍
cwlite Device
gain = 
    mode = high
    gain = 30
    db   = 24.8359375
adc = 
    state      = False
    basic_mode = rising_edge
    timeout    = 2
    offset     = 0
    presamples = 0
    samples    = 5000
    decimate   = 1
    trig_count = 11872602
clock = 
    adc_src       = clkgen_x4
    adc_phase     = 0
    adc_freq      = 26212486
    adc_rate      = 26212486.0
    adc_locked    = True
    freq_ctr      = 2714057
    freq_ctr_src  = extclk
    clkgen_src    = system
    extclk_freq   = 10000000
    clkgen_mul    = 2
    clkgen_div    = 26
    clkgen_freq   = 7384615.384615385
    clkgen_locked = True
trigger = 
    triggers = tio4
    module   = basic
io = 
    tio1       = serial_rx
    tio2       = serial_tx
    tio3       = high_z
    tio4       = high_z
    pdid       = high_z
    pdic       = low
    nrst       = high_z
    glitch_hp  = False
    glitch_lp  = False
    extclk_src = hs1
    hs2        = clkgen
    target_pwr = True
glitch = 
    clk_src     = target
    width       = 10.15625
    width_fine  = 0
    offset      = 10.15625
    offset_fine = 0
    trigger_src = manual
    arm_timing  = after_scope
    ext_offset  = 0
    repeat      = 1
    output      = clock_xor

This has resolved the last connection issue i was originally seeing!   However, i am still not able to connect.  is there any recommended scope setting to float the TDO pin or something to do with IR?

make upload
scripts/upload --hex /home/nlivic/rv2/freedom-e-sdk/software/hello/debug/hello.hex --jlink JLinkExe
SEGGER J-Link Commander V6.70 (Compiled Mar 27 2020 14:37:00)
DLL version V6.70, compiled Mar 27 2020 14:36:47

Connecting to J-Link via USB...O.K.
Firmware: J-Link V11 compiled Mar 19 2020 11:11:08
Hardware version: V11.00
S/N: 601001907
License(s): RDI, FlashBP, FlashDL, JFlash, GDB
VTref=3.328V
Device "FE310" selected.

Connecting to target via JTAG
ConfigTargetSettings() start
ConfigTargetSettings() end
Could not measure total IR len. TDO is constant high.
Cannot connect to target.
J-Link>Target connection not established yet but required for command.
Device "FE310" selected.

colinoflynn, 2020-03-30:
>  i had actually executed another block which had nrst high when trying j-link connection.

This is one of the common gotchas with Jupyter.... sometimes it's useful to reset everything (kernel etc) and do a straight run-through!

I'm going to close this & we'll add some documentation as well around the nrst issue.

colinoflynn, 2020-03-30:
Oops missed second issue was still there.

colinoflynn, 2020-03-30:
The TDO pins don't route anywhere except the JTAG header. They 'should' be all correct here. It can be useful to reduce the JTAG clock frequency sometimes as the default can be pretty fast.

I've tested the FE310 using J-Flash on Windows as well. Check any soldering issues around the QFN on the FE310 (I assume you're using an early proto board here?). You might find it easier to get help on forum.newae.com as well in case other people are using that beta version.

tonkapango, 2020-03-30:
thanks colin. ack on the TDO pins.  i can't visible see any QFN issues, but this is not my domain expertise, i am assuming there is probably better ways to check this package connect?

yes using early rv prototype target.  i believe you sent a couple to robert at nvidia a few months back i got one of them.  ok cool i will post on the forum and play around with different jtag fequencies.

thanks! ok to close if this thread too.

tonkapango, 2020-03-30:
FYI. looks like forum is only on the old site, but without any posts.

tonkapango, 2020-03-30:
BTW, tried JTAG clk speeds from 4000 KHz (default) down through 1 KHz and the same error happens.

colinoflynn, 2020-05-02:
Think this was covered elsewhere (vcc setting issue), please re-open if not!

closed 2020-05-02 by colinoflynn