newaetech/chipwhisperer #261: Negative values for scope.glitch.offset wrongly interpreted

shrewdnimblesquirrel, 2020-04-08:
Hey guys,

it seems the negative range (-50...-0) is not interpreted so that the glitch is moved to the second half of the clock period as mentioned in the API docs. Instead the glitch is starting as if one had used 30...40 as offset. This seems to be a bug, maybe it is caused by the change to python 3?

alex-dewar, 2020-04-09:
It's pretty unlikely that it's an issue with changing to Python 3, since that part of the code was unchanged from CW4 to CW5. There's definitely a physical difference between the negative and positive offsets, since negative offsets are the only ones that work on certain targets. The glitches also behave the same between CW4 and CW5, i.e. you can use the same negative glitch offset settings and get glitches in both CW4 and CW5.

@colinoflynn might be able to follow up better than I can, since he knows more about the glitch module than I do.

shrewdnimblesquirrel, 2020-04-15:
I checked the [glitch module diagram](https://chipwhisperer.readthedocs.io/en/latest/api.html#chipwhisperer.scopes.OpenADC.glitch) you linked in the other post. As the enable glitch pulse is generated at the falling edge of the clock, I think I could track down the issue now. Since the enable wire is high during the whole clock, it can happen (depending on the values set for glitch width and offset), that the glitch will be cut off at the end and another one injected in the beginning of the glitchstream.

You can find a picture that hopefully shows what I mean. Observing with the oscilloscope, I started two loops. One with negative values (-48...-5) used for the glitch offset and one with positive ones (5...48). Checking the gif files attached, you can see that two things happen when using positive values for glitch offset:

- for higher offset values there will appear a second (unintended) voltage glitch in the beginning
- the higher the offset is set, the intended voltage glitch will become weaker and disappear

I used scope.glitch.repeat 1 for testing and a resistor as load, to prevent measuring errors by different loads throughout the clock.
[image]
[image]
[image]

alex-dewar, 2020-04-17:
That makes a lot of sense, especially factoring in those oscilloscope readings. It would also kinda explain https://github.com/newaetech/chipwhisperer/issues/262 as well as why negative and positive glitch offsets behave so differently. Not sure what we want to do with this going forward, but I can at least update the documentation to better reflect this.

colinoflynn, 2020-05-01:
Finally got some of my equipment back up (had to move around stuff due to covid-19) so could do some quick testing. One thing I confirmed at least is the internal phase offset works OK - as a note if you need to generate timing relative to the clock, an easy way to do so is to output CLKGEN on HS2, and then scope the mosfet input pin at header JP8 (if using CW-Lite):

[image]

I'm going to confirm the second part of this issue now - thought it easier to break up a little first though.

colinoflynn, 2020-05-01:
OK so able to replicate this based on what you describe. In my case I used a positive offset for example:

    scope.glitch.clk_src = "clkgen"
    scope.glitch.output = "glitch_only"
    scope.glitch.width = 48
    scope.glitch.offset = 40
    scope.glitch.trigger_src = "manual"
    scope.glitch.manual_trigger()

Results in this (blue = clock, red = mosfet drive):

[image]

colinoflynn, 2020-05-01:
Anyway if we have an "oldest large bug award" but this is probably a winner. Will see if easy fix - the clock routing stuff is some of the most hairy part of FPGA (in part due to partial reconfig usage, which is done despite Xilinx claiming Spartan 6 can't do it).

On the plus side - the stochastic nature of glitching may mean it's not directly stopping anything from being successful!

colinoflynn, 2020-05-01:
In progress testing a fix. There was a comment in the Verilog code around the enable logic that flagged this:

	//Need to think carefully about which clock to syncronize this too, and
	//which edge. Lots of trouble as different options on outputs & adjustable
	//phase
	always @(negedge dcm1_clk_out) begin
		glitch_next_reg1 <= glitch_next;
		glitch_next_reg <= glitch_next_reg1;
	end
	
Previously it used the source clock - I think it should be using the "delayed" clock to gate the enable flag. Will take a few days to validate as the FPGA build process hasn't been done in a long time (see issue #245), so I'm spinning up infrastructure to simplify this in the future.... but if this is the rpot problem will be an easy fix.

shrewdnimblesquirrel, 2020-05-01:
I agree the flag should be set by dcm1 and removed by dcm2. The latter is, since the output will get cut and turned off too early with the current implementation. Just can't tell about the edges, since I can't look into the Verilog code right now.

colinoflynn, 2020-05-01:
I now see consistent widths on a quick test with CW-Lite - let me know if this works. FYI be sure to power-cycle the ChipWhisperer before reconnecting after you pull (if you were already connected), just to ensure it wipes the FPGA config so it will us the new one. If you're on a CW-Pro I'm doing the bitstream gen as well for that now (the partial reconfig stuff takes a long time to build...)

shrewdnimblesquirrel, 2020-05-06:
Yes, I'm on the CW-Lite.
I know tried to test your fix but I can't see any changes on the oscilloscope. I then installed ISE and swapped the enable flag to the positive edge so the glitch should move in time and eventually disabled the glitch at all, but the chipwhisperer doesn't care about all these changes. So I think the new bitstream doesn't get uploaded for some reason. Do you got any idea what could be wrong?

colinoflynn, 2020-05-06:
You need to force just a .bit file to be loaded if you want that to be loaded, by default it pulls the bitstream from ```chipwhisperer.hardware.firmware.cwlite``` where it's encoded in the Python file.

If you check that file on your system, does it look like the repo (updated date-stamp is easiest to confirm).

The problem if you force the .bit to be loaded - none of the glitch.width or glitch.offset will work anymore. This relies on partial reconfiguration which needs you to build a special dictionary of partial reconfig info (this is the most hairy part). Just changing the .bit won't do anything as you won't have valid partial reconfig information.

I dumped a bunch of info at https://forum.newae.com/t/cw-lite-cw-pro-custom-bitstream-loading-advanced-debug/1918 - I thought it's more likely to be found in the future on the forum. Will eventually update some documentation...

colinoflynn, 2020-05-06:
PS - if glitch.offset and glitch.width seem broken now, it's likely the partial reconfig info is invalid. That happens when you reload the bitstream but ChipWhisperer doesn't realize it's using a different built - the commands from the forum should force ChipWhisperer to enable/disable the features correctly. However this isn't well tested in CW5 as I haven't modified the bitstream in a long time before this...

shrewdnimblesquirrel, 2020-05-12:
Thank you, the issue was on my side (still had an old version of the API installed while trying to use the git repo)
I can confirm the fix works, but with a new (little) problem:
There seems to be some kind of "race condition" when glitch offset is too low (lower than -45). Then dcm1_out and clk seem to be too close and regardless of scope.glitch.repeat being set to 1 only, an additional glitch will be inserted at the end as if the user would have used scope.glitch.repeat 2

mateBANANE, 2020-06-18:
Hi.

I can confirm, that for _glitch.offset_ values in the ranges of [-50:-45] and [45:50]  a double glitch may occur. This double glitch has the same shape as a default _glitch.repeat=2_ glitch. Strangely, this double glitch occurs only sometimes, which gives the impression, that this could be a race condition.

While I was investigating this, I have taken a look in your commit https://github.com/newaetech/chipwhisperer/commit/a46ac824d99c65bbcab440b58182a76474cd3d9f.

In the file **clockglitch_s6.v:136** you synchronize the enable signal to the falling edge of _dcm1_clk_out_.
``` 
	always @(negedge dcm1_clk_out) begin
		glitch_next_reg1 <= glitch_next;
		glitch_next_reg <= glitch_next_reg1;
	end
```
Now, in the **trigger_resync** module, you synchronize the trigger signal to a clock. At the end, in **reg_clockglitch.v:364***, you sync it to the "default" clock.

```
	trigger_resync resync(
	.reset(reset),
	.clk(sourceclk),
	.exttrig(exttrigger),	
	.offset(clockglitch_offset_reg),
	.exttrigger_resync(exttrigger_resync) 
	);
```

Furthermore, the **trigger_resync**  counts the cycles for the _scope.ext_offset_. Now the module enables the _exttrigger_resync_ only if the counted signals are equal to to the _ext_offset_. 
Therefore, the trigger is synced to the "default" clock. Later, when the trigger becomes the enable signal, it is synced to _dcm1_clk_out_. Could this be the problem for the race condition? 

I tried to fix it by using the _dcm1_clk_out_ as clock for the **trigger_resync** module. This change was done by wiring it into the module in **reg_clockglitch.v:366**. But the outcome was the same as before. 
Surely, I am lacking advanced knowledge about FPGAs and the overall design of the Chipwhisperer. But maybe my investigation helps somehow.

colinoflynn, 2020-06-19:
Thanks for the input! I hadn't had a change to dig into this, so your effort & attempts are very much appreciated.

I'm almost 100% sure you are correct on the race condition - I think there were some code comments noting that "we should test this more...".

I think we need an actual cross-clock resync block there, as there is no "one good" clock to use for everything. But will need to write out all the clock domains to ensure don't (once again!) introduce new bugs.

Feel free to drop any other notes if you run into stuff (even if it doesn't work), as it's good to know what other people are seeing. The multiple clock domains that ChipWhisperer inherently has is always tricky....

mateBANANE, 2020-06-19:
Thank you for your kind words. If I have a new idea or tried a new solution, I will post it here.

mateBANANE, 2020-07-01:
Hi,
I had another idea. I tried to sync the enable signal to _dcm1_clk_out_ and _dcm2_clk_out_. I wanted to enable it on negedge of _dcm1_clk_out_ and disable it on the negedge (or posedge) of _dmc2_clk_out_. After seeing that I cannot do that in one always block, due to multiple driver issues, I tried to find different solutions to solve this, which resulted only in the "design is unroutable error".
 
I hope you find another fix for this problem. 

Cheers

alex-dewar, 2022-06-06:
Closing, as I believe https://github.com/newaetech/chipwhisperer/commit/5fd1f2561b8ad9557c14a3da75e0996939589a61 has fixed this.

closed 2022-06-06 by alex-dewar