newaetech/chipwhisperer #262: Scope.glitch.width limited to 50

shrewdnimblesquirrel, 2020-04-08:
Hey folks,

for some reason the glitch width is limited to 50% of the period. Especially when glitching targets with higher complexity (e.g. STM32) this is not sufficient enough in order to inject faults.

Can you please raise this to 100%?

alex-dewar, 2020-04-09:
Looking at the [glitch module diagram](https://chipwhisperer.readthedocs.io/en/latest/api.html#chipwhisperer.scopes.OpenADC.glitch), I don't believe this is possible. The implementation would have to be more complex to make this possible, but again @colinoflynn would probably be able to give a better answer than I can.

One thing that might be worth trying is clocking the ChipWhisperer at a much higher speed than the target and using `enable_only`/repeat to set the width and `ext_offset` as the offset. This is the strategy used for the LPC1114 bootloader glitch and I've also found success with this technique when glitching the STM32F0.

colinoflynn, 2020-05-01:
The glitch width of 50% is a hard limitation of the DCM. As Alex mentioned the general way we use it for longer glitches is with "enable_only", there the "repeat" sets the width. The very narrow glitches are mostly needed for clock glitching... with voltage glitching most targets require longer widths you can only achieve with "enable_only". By adjusting the CLKGEN speed and repeat number you can get very fine-grained control on the actual output width.

closed 2020-05-01 by colinoflynn

colinoflynn, 2020-05-02:
Minor additional note: new FPGA build increases the clock glitch cycles out to 8192 from 255. This should be helpful when using 'enable' mode for even longer glitches without changing the clock frequency.