newaetech/chipwhisperer #268: how to use it?

zhubin16167758, 2020-06-09:


alex-dewar, 2020-06-09:
Hi,

Most of our documentation on the software side is on our [readthedocs page](https://chipwhisperer.readthedocs.io/en/latest/index.html), including installation instructions and the API. The jupyter submodule also has jupyter notebook based tutorials, completed versions of which can be found on our readthedocs page: https://chipwhisperer.readthedocs.io/en/latest/tutorials.html.

We also have [forums](https://forum.newae.com), which is probably a better place to ask more general questions like this.

Alex

colinoflynn, 2020-06-19:
I'm going to close this issue - but might actually pin a new "getting started" issue since the documentation is for now easy to miss some important links! If we can save people some time makes sense.

closed 2020-06-19 by colinoflynn

JohnLoveMm, 2020-06-24:
I'm glad you answered my question,thank you very much.