newaetech/chipwhisperer #272: CWLITE Externally Triggered hp_glitch stays high after trigger

hypoxic, 2020-07-02:
It appears the glitch_hp stays asserted indefinitely with this configuration and positive 'scope.glitch.width'

To reproduce add the following to Fault_2-Introduction_to_Vcc_Glitch_Attacks.ipynb in section **1.3.4**. You'll also have to hardwire section 1.3.2 to get the "glitch_on" functions with an CWLITEXMEGA

`#clock
scope.clock.clkgen_src = "system"
scope.clock.clkgen_freq = 59076942  

while not scope.clock.clkgen_locked:
    time.sleep(0.1)
scope.adc.timeout = 0.5

#trigger
scope.trigger.triggers = 'tio4'
scope.clock.adc_src = "clkgen_x1"
scope.clock.freq_ctr_src = 'clkgen'
scope.adc.basic_mode = 'rising_edge'
while not scope.clock.adc_locked:
    time.sleep(0.5)

#scope.adc.basic_mode = 'falling_edge' # Does not work

#glitch control
scope.glitch.arm_timing = 'after_scope'
scope.glitch.clk_src = "clkgen"
scope.glitch.trigger_src = 'ext_single'
scope.glitch.output = "glitch_only"
scope.glitch.repeat = 1

scope.io.target_pwr = True
scope.io.hs2 = "clkgen"

scope.glitch.offset_fine = 0
scope.glitch.repeat = 1`

alex-dewar, 2020-07-03:
Potentially related to https://github.com/newaetech/chipwhisperer/issues/90?

Once the glitch sticks high, if you turn the high power glitch off and on, does the glitch stay off?

alex-dewar, 2020-11-06:
Was finally about to try this out, and I believe this case is just a combination of the set frequency on the XMega being well above it's max frequency and the shunt resistor dropping too much voltage at higher frequencies.

closed 2020-11-06 by alex-dewar

hypoxic, 2020-11-06:
Ok. I wasn't actually attached to a XMEGA, just used that to get the sample setup. 
What does the shunt resistor have to do with the HV glitcher?

alex-dewar, 2020-11-09:
Whoops sorry. Basically, as you increase the frequency on the target, its power draw increases. This means that the voltage drop across the shunt resistor increases and the Vcc that the chip sees is lower. With these settings, I was seeing about 2V at the chip, which is in spec for low frequencies, but not for higher frequencies (that being said, I believe the max frequency the XMega can run at is 32MHz).  The chip still kind of works in this situation, but it's very inconsistent.