newaetech/chipwhisperer #278: running cw software in linux

mati335, 2020-09-01:
so sorry for the noob question but how do you run the analyzer and capture software without jupyter

alex-dewar, 2020-09-01:
Almost exactly the same as within Jupyter, via the ChipWhisperer API, minus the Jupyter specific magics (like `%matplotlib notebook` and `%%bash` and the like).

If you're wondering about the old GUIs, those were removed with the ChipWhisperer 5 update.

colinoflynn, 2020-09-01:
Also - the old GUIs were removed in part as they relied on PySide (not PySide2), which is Python2.7 only and it was non-trivial to port, so it's not a case of them just being somewhere hidden!

If you need the install directions, you can see them at https://chipwhisperer.readthedocs.io/en/latest/prerequisites.html#prerequisites-linux

closed 2020-09-01 by colinoflynn