newaetech/chipwhisperer #279: floating-point operation with STM32F4

mcsmonk, 2020-09-02:
Hi

My below code is not working with original makefile.stm32f4
I am using it after modifying makefile to run for experimentation, but I am curious if there is any problem.

- Environments
  - CW-Pro / CW308-STM32F405-R03
  - CW version 5.2.1 in virtual box

``` C
// simpleserial-floatingpoint.c

uint8_t get_pt(uint8_t* pt)
{
	float *fpt = (float *)pt;

	trigger_high();

	fpt[0] = fpt[0] * fpt[0];
	fpt[1] = fpt[1] * fpt[0];
	fpt[2] = fpt[2] * fpt[2];
	fpt[3] = fpt[3] * fpt[3];

	trigger_low();
	
	simpleserial_put('r', 16, pt);
	return 0x00;
}
```

``` SH
# hal/stm32f4/Makefile.stm32f4

#CFLAGS += -mthumb -mfloat-abi=soft -mfpu=fpv4-sp-d16 -fmessage-length=0 -ffunction-sections
#CPPFLAGS += -mthumb -mfloat-abi=soft -mfpu=fpv4-sp-d16 -fmessage-length=0 -ffunction-sections
#ASFLAGS += -mthumb -mfloat-abi=soft -mfpu=fpv4-sp-d16 -fmessage-length=0 -ffunction-sections
CFLAGS += -mthumb -mfloat-abi=soft -ffunction-sections
CPPFLAGS += -mthumb -mfloat-abi=soft -ffunction-sections
ASFLAGS += -mthumb -mfloat-abi=soft -ffunction-sections
```

alex-dewar, 2020-09-02:
Thanks for bringing this up! 

Hardware floating point has a tendency to break the build/binary depending on what `arm-none-eabi-gcc` version is, so I'm guessing this was intended to be a change from hardware to software floating point anyway...

We should probably disable hardware floating point across the board since the support is so spotty with `arm-none-eabi-gcc` and just rely on software floating point, making a note in the target documentation. If people want to study hardware floating point, they can manually enable it.

I'll get this fixed up, then close this issue.

alex-dewar, 2020-09-02:
HW floating point disabled by 8df5df0546a557fe2af1213f7c9890602358a134

closed 2020-09-02 by alex-dewar