newaetech/chipwhisperer #281: Problem in Jetbrains PyCharm

noeyheadb, 2020-09-11:
Hi!

I used it well until 2 months ago at Jetbrains PyCharm.

But suddenly, the following error appears when using the chipwhisperer module in the PyCharm.

Code: 
```python
import chipwhisperer as cw
scope = cw.scope(scope_type=cw.scopes.OpenADC)
target = cw.target(scope)        # <<< error occured (after print 'Serial baud rate = 38400')
scope.default_setup()
ktp = cw.ktp.Basic()
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
```

Error:
```
Traceback (most recent call last):
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 495, in cmdReadMem
    data = self.readCtrl(cmd, dlen=dlen)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 471, in readCtrl
    return self.usbdev().ctrl_transfer(0xC1, cmd, value, 0, dlen, timeout=self._timeout)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb0.py", line 593, in ctrl_transfer
    timeout
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xba\xb7\xb9\xb5\xe5 \xc1\xbe\xb7\xe1 \xb6\xc7\xb4\xc2 \xc0\xc0\xbf\xeb \xc7\xc1\xb7\xce\xb1\xd7\xb7\xa5 \xbf\xe4\xc3\xbb \xb6\xa7\xb9\xae\xbf\xa1 I/O \xc0\xdb\xbe\xf7\xc0\xcc \xc3\xeb\xbc\xd2\xb5\xc7\xbe\xfa\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
Traceback (most recent call last):
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 489, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 464, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb0.py", line 593, in ctrl_transfer
    timeout
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xba\xb7\xb9\xb5\xe5 \xc1\xbe\xb7\xe1 \xb6\xc7\xb4\xc2 \xc0\xc0\xbf\xeb \xc7\xc1\xb7\xce\xb1\xd7\xb7\xa5 \xbf\xe4\xc3\xbb \xb6\xa7\xb9\xae\xbf\xa1 I/O \xc0\xdb\xbe\xf7\xc0\xcc \xc3\xeb\xbc\xd2\xb5\xc7\xbe\xfa\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
Traceback (most recent call last):
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\IPython\core\interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-3-a20dde5ef0b0>", line 1, in <module>
    cw.capture_single(True, update_trig_cnt=True)
  File "C:\Users\511-4\Desktop\PROJECT DIR\cw-lite-wrapper\cw_lite_wrapper\cw_lite_wrapper.py", line 146, in capture_single
    trace = self._capture()
  File "C:\Users\511-4\Desktop\PROJECT DIR\cw-lite-wrapper\cw_lite_wrapper\cw_lite_wrapper.py", line 130, in _capture
    trace = cw.capture_trace(self._scope, self._target, pt, key)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\__init__.py", line 285, in capture_trace
    ret = scope.capture()
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\capture\scopes\OpenADC.py", line 321, in capture
    return self.qtadc.capture(self.adc.offset, self.clock.adc_freq, self.adc.samples)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\capture\scopes\_qt.py", line 111, in capture
    timeout = self.sc.capture(offset, adc_freq, samples)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1876, in capture
    status = self.getStatus()
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1680, in getStatus
    result = self.sendMessage(CODE_READ, ADDR_STATUS)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1556, in sendMessage
    data = bytearray(self.serial.cmdReadMem(address, datalen))
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 693, in cmdReadMem
    return self.usbseralizer.cmdReadMem(addr, dlen)
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 201, in cmdReadMem
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "C:\Users\511-4\AppData\Local\Programs\Python\Python37\lib\site-packages\chipwhisperer\hardware\naeusb\naeusb.py", line 154, in process_rx
    raise payload
usb.core.USBError: [Errno None] None
```

However, after downloading and installing `Windows.Installer.Chipwhisperer.v5.3.0.Setup.64-bit.exe`, run the same code in jupyter and it will work well.

Previously, I used chipwhisperer version 5.1.3 in PyCham.

However, neither version 5.1.3 nor 5.3.1 currently works with the above error.

What's the problem?

alex-dewar, 2020-09-11:
Hi,

The Windows installer packages its own copy of Python to use. Which Python interpreter are you using with PyCharm?

noeyheadb, 2020-09-11:
PyCharm used `UserHome/ChipWhisperer5_64/WPy64-3740/python-3.7.4.amd64/python.exe`.
 (installed from `Windows.Installer.Chipwhisperer.v5.3.0.Setup.64-bit.exe`)

The above error occurs only in pycharm when using the same interpreter used by the jupyter.

alex-dewar, 2020-09-11:
Is it possible that you're connected to the ChipWhisperer in two python instances at the same time? If you unplug+replug the ChipWhisperer and rerun in PyCharm, does the error still occur?

alex-dewar, 2020-09-11:
I'm not able to replicate on my end using either my usual native Python interpreter, or the one from the ChipWhisperer installer. This might be some oddity with the libusb0 driver.

noeyheadb, 2020-09-14:
I tried again right after rebooting the computer, and similarly the following error appears:
```
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32
import chipwhisperer as cw
scope = cw.scope(scope_type=cw.scopes.OpenADC)
WARNING:root:Your firmware is outdated - latest is 0.20. Suggested to update firmware, as you may experience errors
See https://chipwhisperer.readthedocs.io/en/latest/api.html#firmware-update
target = cw.target(scope)
Serial baud rate = 38400
Traceback (most recent call last):
Traceback (most recent call last):
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 495, in cmdReadMem
    data = self.readCtrl(cmd, dlen=dlen)
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 495, in cmdReadMem
    data = self.readCtrl(cmd, dlen=dlen)
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 471, in readCtrl
    return self.usbdev().ctrl_transfer(0xC1, cmd, value, 0, dlen, timeout=self._timeout)
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 593, in ctrl_transfer
    timeout
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xc3\xbd\xba\xc5\xdb\xbf\xa1 \xba\xce\xc2\xf8\xb5\xc8 \xc0\xe5\xc4\xa1\xb0\xa1 \xc0\xdb\xb5\xbf\xc7\xcf\xc1\xf6 \xbe\xca\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 471, in readCtrl
    return self.usbdev().ctrl_transfer(0xC1, cmd, value, 0, dlen, timeout=self._timeout)
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 593, in ctrl_transfer
    timeout
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xc3\xbd\xba\xc5\xdb\xbf\xa1 \xba\xce\xc2\xf8\xb5\xc8 \xc0\xe5\xc4\xa1\xb0\xa1 \xc0\xdb\xb5\xbf\xc7\xcf\xc1\xf6 \xbe\xca\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
Traceback (most recent call last):
  File "C:\Users\511-4\AppData\Local\JetBrains\Toolbox\apps\PyCharm-P\ch-0\202.6948.78\plugins\python\helpers\pydev\_pydevd_bundle\pydevd_thrift.py", line 305, in var_to_struct
    value = format % v
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 361, in __str__
    return self.__repr__()
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 354, in __repr__
    ret = "%s Device\n" % self._getCWType()
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 198, in _getCWType
    hwInfoVer = self.qtadc.sc.hwInfo.versions()[2]
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 80, in versions
    result = self.oa.sendMessage(CODE_READ, ADDR_VERSIONS, maxResp=6)
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1556, in sendMessage
    data = bytearray(self.serial.cmdReadMem(address, datalen))
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 693, in cmdReadMem
    return self.usbseralizer.cmdReadMem(addr, dlen)
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 201, in cmdReadMem
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 154, in process_rx
    raise payload
usb.core.USBError: [Errno None] None
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "c:\users\511-4\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 296, in txrx
    response = self.usbdev().ctrl_transfer(payload[0], payload[1], payload[2], payload[3], payload[4], timeout=self._timeout)
  File "C:\Users\511-4\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
Traceback (most recent call last):
[... cut ...] c\xd2\xb5\xc7\xbe\xfa\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
```

And I tried it on another PC that worked well.
PyCharm's first attempt worked without error. However, the same error occurs when the Python interpreter is shut down and tried again. (Even if you reboot and try immediately)

However, it works well without errors, if i still use the jupyter, 

If there is a problem with the driver, the same error should occur on the jupyter or other PC. 
But that's not the case, so I don't know what to do.

noeyheadb, 2020-09-14:
The same error occurs when i shut down all other Python interpreter.

alex-dewar, 2020-09-14:
Yeah, I'm pretty stumped on this one as well. There's USB transfers when you connect to the scope, so I don't think it's USB communication in general not working. If you swap `scope.default_setup()` and `cw.target(scope)`, does the error occur during `scope.default_setup()` instead?

We've also recently updated the USB firmware on the ChipWhisperer. Have you updated to the latest firmware? We added a command to check how much data is in the tx_buffer. We do check you have the required firmware version, but it might be that this check is failing somehow?

alex-dewar, 2020-09-15:
Looking again, I can see that your ChipWhisperer firmware is out of date. Can you try updating it and seeing if that fixes the issue?

noeyheadb, 2021-01-27:
I'm sorry for not responding for a long time.

I couldn't do it because I had difficulty updating the cw-lite firmware. (COM port problem)
(My cw is still 0.11 ver.)

After reviewing this problem again, I found that the problem only occurs when using the interactive console of PyCharm scientific mode. 

When I run the Python interactive console in scientific mode mode, it runs with the following command.

```
"D:\Projects\PycharmProjects\Virtualenv Environments\chipwhisperer-venv\Scripts\python.exe" "C:\Users\Daehyeon Bae\AppData\Local\JetBrains\Toolbox\apps\PyCharm-P\ch-0\203.6682.179\plugins\python\helpers\pydev\pydevconsole.py" --mode=client --port=49543
```

When I run the scripts normally instead of interactive console in sci. mode, it also works well in Pycharm.

I think there is a conflict between sci. mode of pycharm and Chipwhisperer.

[image]
(Scientific mode of pycharm)

noeyheadb, 2021-01-27:
And I have one more question about firmware update.

I tried connecting cw to all my USB ports, but I couldn't find any COM ports.

It is always connected to a port such as "Port_#0007.Hub_#0002". 

Therefore, when updating the firmware, typing "Port_#0007.Hub_#0002" as the identifier(not COM#) of the port results in an error.

 What should I do for firware update?

alex-dewar, 2021-01-27:
That's strange, it should always show up as a COM Port, assuming the firmware's actually been erased. Are you still able to connect to the ChipWhisperer? If so, the firmware isn't actually erased.

Assuming the firmware isn't erased and you're on the newest ChipWhisperer software, you should be able to use the `auto_program()` method of the `SAMFWLoader` to automatically erase and detect the COM port.

noeyheadb, 2021-01-28:
Thanks. I updated the firmware to the latest version using `auto_program()`.

However, the error in pycharm scientific mode as above still occurs.

After trying a few more, I found the following:
- Sci. mode doesn't always cause an error. Sometimes it works well.
- If I run the code line by line, the probability of error is high.
- If I run the code block by block, the probability of error seems to be around 50%.
- If I run the entire code at the same time, the probability of error are not high.

noeyheadb, 2021-02-01:
Yes.. Even with the latest version of the driver, the exception still occurs.

As mentioned before, if the connection code and the capture code are executed at the same time as `In[2]`, an error does not occur.
However, if the connection code and the capture code are not executed at the same time, an error occurs with a high probability. (e.g. Run `In[2]` -> `In[3]` separately)

```
IPython 7.7.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 7.7.0
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32

In[2] : 
import chipwhisperer as cw
scope = cw.scope()
scope.default_setup()
target = cw.target(scope)
ktp = cw.ktp.Basic()
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stdout : 
Serial baud rate = 38400
Trace(wave=array([ 0.10058594, -0.171875  , -0.16796875, ...,  0.0234375 ,
        0.04589844, -0.00097656]), textin=CWbytearray(b'94 f0 d1 0a 78 55 df 68 6c 01 ec 9a 9d 7f 88 f6'), textout=CWbytearray(b'2f 2d 85 ba 97 2e fb 63 f8 73 a0 98 19 76 e1 9b'), key=CWbytearray(b'2b 7e 15 16 28 ae d2 a6 ab f7 15 88 09 cf 4f 3c'))
        
In[3] : 
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stderr : 
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 506, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 481, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 609, in ctrl_transfer
    timeout
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xc3\xbd\xba\xc5\xdb\xbf\xa1 \xba\xce\xc2\xf8\xb5\xc8 \xc0\xe5\xc4\xa1\xb0\xa1 \xc0\xdb\xb5\xbf\xc7\xcf\xc1\xf6 \xbe\xca\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 506, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 481, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 609, in ctrl_transfer
    timeout
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xc3\xbd\xba\xc5\xdb\xbf\xa1 \xba\xce\xc2\xf8\xb5\xc8 \xc0\xe5\xc4\xa1\xb0\xa1 \xc0\xdb\xb5\xbf\xc7\xcf\xc1\xf6 \xbe\xca\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
Traceback (most recent call last):
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\IPython\core\interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-3-507bbb1f1242>", line 2, in <module>
    trace = cw.capture_trace(scope, target, pt, key)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\__init__.py", line 320, in capture_trace
    ret = scope.capture()
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 325, in capture
    return self.qtadc.capture(self.adc.offset, self.clock.adc_freq, self.adc.samples)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_qt.py", line 110, in capture
    timeout = self.sc.capture(offset, adc_freq, samples)
[... 38 lines cut ...]
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1573, in sendMessage
    data = bytearray(self.serial.cmdReadMem(address, datalen))
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 710, in cmdReadMem
    return self.usbseralizer.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 201, in cmdReadMem
[... cut ...]
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 609, in ctrl_transfer
    timeout
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xbd\xc3\xbd\xba\xc5\xdb\xbf\xa1 \xba\xce\xc2\xf8\xb5\xc8 \xc0\xe5\xc4\xa1\xb0\xa1 \xc0\xdb\xb5\xbf\xc7\xcf\xc1\xf6 \xbe\xca\xbd\xc0\xb4\xcf\xb4\xd9.\r\n\n'
```

alex-dewar, 2021-02-01:
This appears to still be using the old version of the driver. Does your ChipWhisperer enumerate as a WinUSB device (IIRC the old version is a libusb32 device)? If not, you're still on the old version, as can be seen by the use of `libusb0.py`. Can you confirm that your device enumerates as a WinUSB device?

If you do have the correct driver installed, something might be going wrong in the process that selects which driver to use.

alex-dewar, 2021-02-03:
I should mention that you can see what your device has enumerated as by going to device manager in Windows.

noeyheadb, 2021-02-04:
After installing using Chipwhisperer.v5.4.0.Setup.64-bit.exe, I manually installed the driver included in the package(zip file) with the device manager and used it. As a result, cw-lite was recognized as a libusb32 device.  -> use libusb0.py

However, I removed the driver from device manager, and it is recognized as WinUSB device.
If it is recognized as WinUSB device, is it considered to have the latest driver installed?

After being recognized as a winusb device, it seems that libusb1.py is used instead of libusb0.py.

## libusb32-device
[image]

## WinUSB device
[image]

alex-dewar, 2021-02-04:
Yup, WinUSB is the correct driver now and libusb1.py is the correct backend. Do you still see errors?

noeyheadb, 2021-02-05:
An error occurs even when using the latest driver and cwlite firmware (0.23)...

However, the same error as before does not occur, and a 'pipe error' occurs.

And again, jyupyter does not cause an error, only in PyCharm interactive interpreter.

```
In[2] : 
import chipwhisperer as cw
scope = cw.scope()
scope.default_setup()
target = cw.target(scope)
ktp = cw.ktp.Basic()
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stdout :
Serial baud rate = 38400
Trace(wave=array([ 0.10742188, -0.16210938, -0.15917969, ...,  0.02441406,
        0.04492188,  0.00097656]), textin=CWbytearray(b'ff de ba 77 6f 29 61 52 a0 99 50 8a 91 c0 08 4d'), textout=CWbytearray(b'75 fa 33 c0 3d 23 70 d3 86 a8 c2 f6 be f6 ac 18'), key=CWbytearray(b'2b 7e 15 16 28 ae d2 a6 ab f7 15 88 09 cf 4f 3c'))
        
In[3]:
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stderr: 
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 506, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 481, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
Traceback (most recent call last):
  File "C:\Users\Daehyeon Bae\AppData\Local\JetBrains\Toolbox\apps\PyCharm-P\ch-0\203.7148.72\plugins\python\helpers\pydev\_pydevd_bundle\pydevd_thrift.py", line 305, in var_to_struct
    value = format % v
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 413, in __str__
    return self.__repr__()
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 406, in __repr__
    ret = "%s Device\n" % self._getCWType()
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 208, in _getCWType
    hwInfoVer = self.qtadc.sc.hwInfo.versions()[2]
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 80, in versions
    result = self.oa.sendMessage(CODE_READ, ADDR_VERSIONS, maxResp=6)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1661, in sendMessage
    data = bytearray(self.serial.cmdReadMem(address, datalen))
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 710, in cmdReadMem
    return self.usbseralizer.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 201, in cmdReadMem
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 154, in process_rx
    raise payload
usb.core.USBError: [Errno None] 32
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 506, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 481, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

stdout :
Trace(wave=array([ 0.10253906, -0.16210938, -0.15917969, ...,  0.02929688,
        0.05273438,  0.00683594]), textin=CWbytearray(b'0e 1d ca 52 cd 05 1a 95 5e eb 0f b3 72 96 42 0b'), textout=CWbytearray(b'de e4 41 82 fb b5 b5 b8 8d 7a 0b 15 b3 cf da 75'), key=CWbytearray(b'2b 7e 15 16 28 ae d2 a6 ab f7 15 88 09 cf 4f 3c'))
```

alex-dewar, 2021-02-05:
Interestingly, that third code block there indicates that execution was successful (i.e. we got a trace + plaintext back and execution wasn't interrupted permanently). If you rerun that block, do you still get new traces back?

noeyheadb, 2021-02-06:
It works in three different ways randomly.
- Return trace normally without exception (`In[3]`, `In[5]`)
- Return trace with exception (`In[6]`)
- Throw exception without returning trace (`In[2]`, `In[4]`)

And sometimes 'float division by zero' exception occurs like `In[2]`, not pipe error.

```
In[2] :
import chipwhisperer as cw
scope = cw.scope()
scope.default_setup()
target = cw.target(scope)
ktp = cw.ktp.Basic()
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stdout :
Serial baud rate = 38400

stderr :
Traceback (most recent call last):
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\IPython\core\interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-88319a30740c>", line 7, in <module>
    trace = cw.capture_trace(scope, target, pt, key)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\__init__.py", line 320, in capture_trace
    ret = scope.capture()
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 335, in capture
    return self.qtadc.capture(self.adc.offset, self.clock.adc_freq, self.adc.samples)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_qt.py", line 110, in capture
    timeout = self.sc.capture(offset, adc_freq, samples)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 2020, in capture
    cap_delay = (7.37E6 * 4 * samples) / (adc_freq * 24400)
ZeroDivisionError: float division by zero

In[3] :
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stdout :
Trace(wave=array([ 0.10644531, -0.171875  , -0.16503906, ...,  0.015625  ,
        0.0390625 , -0.00488281]), textin=CWbytearray(b'8c 88 96 c0 92 6f d9 67 8b fc 10 32 a5 35 7d 8c'), textout=CWbytearray(b'cc 83 25 a1 23 33 cf bf 79 f7 fd 00 0d 1b 71 8c'), key=CWbytearray(b'2b 7e 15 16 28 ae d2 a6 ab f7 15 88 09 cf 4f 3c'))
        
In[4] :
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)

stderr :
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 506, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 481, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
Traceback (most recent call last):
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\IPython\core\interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-4-507bbb1f1242>", line 2, in <module>
[... 73 lines cut ...]
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 201, in cmdReadMem
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 154, in process_rx
    raise payload
usb.core.USBError: [Errno None] 32
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 506, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 481, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

stdout :
Trace(wave=array([ 0.10644531, -0.16699219, -0.16113281, ...,  0.02148438,
        0.04492188, -0.00195312]), textin=CWbytearray(b'1b 9b 3f b2 01 5e b8 40 fd af 3b 9b 67 00 21 ec'), textout=CWbytearray(b'f4 06 16 36 34 45 bd 18 8f 87 2a 78 5d cd c8 10'), key=CWbytearray(b'2b 7e 15 16 28 ae d2 a6 ab f7 15 88 09 cf 4f 3c'))
```

alex-dewar, 2021-02-10:
The only thing I can think of is that the USB communication is being interrupted by PyCharm which is causing USB communication errors. You can see something similar if you interrupt capture with something like Ctrl-C.

alex-dewar, 2021-03-26:
I was able to track down an fix one pipe error. Can you try the latest firmware and let me know if you still run into a pipe error?

noeyheadb, 2021-03-26:
I still get an pipe error in only PyCharm after updating the firmware(0.3) and CW_sw(33620d6) 😥   
However, I am using it well in jupyter, so there is no problem. Can I close the issue?

_**In[1]**_ :
```
import chipwhisperer as cw
scope = cw.scope(scope_type=cw.scopes.OpenADC)
target = cw.target(scope)
scope.default_setup()
```

_**Out[1]**_ :
```
Serial baud rate = 38400
```

_**In[2]**_ : 
```
ktp = cw.ktp.Basic()
key, pt = ktp.new_pair()
trace = cw.capture_trace(scope, target, pt, key)
print(trace)
```

_**Stderr[2]**_ :
```
WARNING:root:Attempting pipe error fix - typically safe to ignore
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 499, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 473, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 504, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 473, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
WARNING:root:Attempting pipe error fix - typically safe to ignore
WARNING:root:Attempting pipe error fix - typically safe to ignore
Traceback (most recent call last):
  File "C:\Users\Daehyeon Bae\AppData\Local\JetBrains\Toolbox\apps\PyCharm-P\ch-0\203.7717.65\plugins\python\helpers\pydev\_pydevd_bundle\pydevd_thrift.py", line 305, in var_to_struct
    value = format % v
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 417, in __str__
    return self.__repr__()
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 410, in __repr__
    ret = "%s Device\n" % self._getCWType()
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py", line 212, in _getCWType
    hwInfoVer = self.qtadc.sc.hwInfo.versions()[2]
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 80, in versions
    result = self.oa.sendMessage(CODE_READ, ADDR_VERSIONS, maxResp=6)
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py", line 1661, in sendMessage
    data = bytearray(self.serial.cmdReadMem(address, datalen))
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 739, in cmdReadMem
[... 243 lines cut ...]
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 473, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "c:\users\daehyeon bae\chipwhisperer5_64\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py", line 304, in txrx
    response = self.cmdReadMem(addr, dlen)
[... cut ...]
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "C:\Users\Daehyeon Bae\ChipWhisperer5_64\WPy64-3740\python-3.7.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
```

alex-dewar, 2021-03-26:
Hmm, yeah it looks like this is a different situation causing the pipe error. I'm pretty stumped and, unfortunately, I don't think I'll be able to make much progress without being able to recreate it and attach a USB sniffer/debugger.

Feel free to close the issue. I've updated the title to indicate that this is a pipe error to hopefully make it a little easier to find if someone runs into the same issue.

Thanks for your patience while I tried to work through the issue!

noeyheadb, 2021-03-26:
Thank you for your help! 
I hope this problem will be solved someday :)

closed 2021-03-26 by noeyheadb