newaetech/chipwhisperer #286: ChipWhisperer < 5.0 for Python3

ryancor, 2020-09-28:
Is there a way to get python3 support for the older GUI versions? I'll be honest, ever since the update I do not understand a single thing on how to get this to work in the Jupyter notebook. The GUI was so easy to understand in the earlier versions. I've spent months trying to understand these new updates but they are driving me a little crazy, and I want to be able to use this product on my Ubuntu20 VM but it doesn't seem like i'll be able to

alex-dewar, 2020-09-28:
Unfortunately, the answer is probably no. It looks like PySide, the library we used for, supports up to Python 3.4 on non-Windows platforms, so if you can get a Python version that old, it might be upgradable. Otherwise, it's been replaced by PySide 2, which runs off of Qt5 instead of Qt4. I'm not how much work it would be moving from PySide/Qt4 to PySide2/Qt5, but I'd guess it would be substantial. 

If you've got any questions about using CW5, we'll be happy to help out.

ryancor, 2020-09-28:
I'm running python3 -m notebook inside the CW directory. What I'm confused about is how to even get the CWCapture and CWAnalyze window?

alex-dewar, 2020-09-28:
To start with, if you haven't cloned the Jupyter submodule (chipwhisperer/jupyter is empty), you'll want to do that by running `git submodule update --init jupyter` in the chipwhisperer directory.

`python3 -m notebook` should open a tab in your browser. When it does, navigate in the tab to the jupyter folder. You should see `0 - Introduction to Jupyter Notebooks.ipynb`, which is our introduction to Jupyter notebooks. I'd recommend checking out `courses/sca101` next, which has adaptations of a lot of the old basic tutorials from the wiki. 

CW5 works a bit differently than previous versions. Instead of the GUI, you can interact with the hardware using the Python API. For example, instead of running the connect script, we run the following commands:

```python
import chipwhisperer as cw
scope = cw.scope()
```

and instead of hitting the capture button, the individual steps are done:

```python
scope.arm()
target.simpleserial_write('p', plaintext)
scope.capture()
```

The same stuff is happening, it's just a bit more explicit now. If you're having trouble running the notebooks, we've got completed versions hosted at: https://chipwhisperer.readthedocs.io/en/latest/tutorials.html#tutorials-sca101. It's all just Python as well, so in the worst case you can run all this stuff in Python as well (well, besides the %%bash blocks)

This is a pretty broad post, so feel free to ask more questions.

colinoflynn, 2020-09-28:
For more context on dropping the GUI - we used a *lot* of special features, code hacks, etc to get some stuff we needed in the gui with PySide. So we can't easy "port" most of it to PySide2 (which is needed for Python3).

TBH I do still keep a Python2 alive with the GUI installed here (the shame!) for some work. We had on the TODO list building a "light-weight GUI" which would run on Python3 - mostly just for viewing power traces in rea-time and stuff that isn't as smooth inside of Jupyter. But it was going to be easier to rebuild on PySide2 the new GUI than it was to port the old stuff to PySide2. There was a community fork that started some of the porting, but also ran into the same conclusion...

Note you can do real-time plotting and similar in Jupyter. Thus for the setup phase, you can still get some of that interactive feel. We're working on some improved docs to show this use-case too. The new system is *so flexible* that it's a lot harder to quickly spin it up, but it also gets rid of a *lot* of limitations we'd see people run into with the GUI.

ryancor, 2020-09-30:
@colinoflynn i completely understand! I think im just so new to jupyter, that im so lost in the documentation and how use a GUI in a notebook lol. It's why I still use version 4.1 of chipwhisperer on my windows machine :D

alex-dewar, 2021-08-20:
Hi,

There's been no activity on this issue for a while, so I'm closing it.

If the issue still exists, let me know and I'll reopen this issue.

closed 2021-08-20 by alex-dewar