newaetech/chipwhisperer #287: Bricked the Nano

osok, 2020-09-30:
I have a Chipwhisperer Nano.

I downloaded the VirtualBox 5.3 VM.
Logged in an got Jupyter running.
Logged into Jupyter.

Ran:
`
import chipwhisperer as cw
scope = cw.scope()
`
That returned the following 
`
WARNING:root:Your firmware is outdated - latest is 0.20. Suggested to update firmware, as you may experience errors
See https://chipwhisperer.readthedocs.io/en/latest/api.html#firmware-update
`

So I followed the instructions there and ran 
`
import chipwhisperer as cw
scope = cw.scope()
programmer = cw.SAMFWLoader(scope=scope)
programmer.enter_bootloader(really_enter=True)
`

The blinking stopped.  I unplugged the CWN from my computer and plugged it back in.  I checked the linux and there were no /dev/ttyUSB# or /dev/ttyACM#.  

So I was not going to be able to run the next commands.  I even tried to attach the device identified in the next part to the VM, still not ttyACM or ttyUSB.

So I checked Windows and I did see a COM17 AT91 USB to Serial Converter, which goes away when I unplug the CWN.

So I launched BOSSA and provided is COM17 and it said it was an unsupported device.

I did try shorting the erase flash pins.   No Change.

So first why did that go so wrong?  Why am I see that verus the BOSSA device?

Second,  I see the programming pins on the 20 pin port, can use a programmer, I have a JLINK and PEMicro Multiportlink Universal to program it using the bin identified on the site?

I see that the 20 pin header
Pin 5 Reset
Pin 7 MISO
Pin 9 MOSI
Pin 11 SCK
2 & 3 are Gnd and Vcc.

Thanks,

alex-dewar, 2020-09-30:
Hi,

1. What's your host operating system? 
1. Can you try https://rtfm.newae.com/Capture/ChipWhisperer-Lite/#linux-usbserial-module-workaround and see if that gets it to show up
1. Try adding your user to the dialout group

osok, 2020-10-01:
So I did notice a behavior that I over looked.  With the VM up, and the CWN unplugged, I plug it in and then select the device AT91 USB to Serial Converter in the VM Devices, which does add the device to the VM.  It is now /dev/ttyACM0.

Then when I run:

`import chipwhisperer as cw`
`programmer = cw.SAMFWLoader(scope=None)`
`programmer.program(<port>, hardware_type='cwnano')`

Nothing happens.   I don't get an error this time.  Last night I was getting an error that the port didn't exist.  If I unplug the device and plug it back in and add the device it still is not recogonized.

alex-dewar, 2020-10-01:
Does the programming say that it's successfully completed? It should print some stuff out when you call `program()`

osok, 2020-10-01:
no

osok, 2020-10-01:
Just tried again and I no longer see the /dev/ttyACM0.  I am noticing something else.  If I unplug it and plug it in then select it to connect to the VM, occasionally the VM crashes and I have to restart it.

So many of my attempts are with a fresh restart of the VM and after connection I cannot see the device in `lsusb` or `ls -l /devACM*`

alex-dewar, 2020-10-01:
I just checked and bossa doesn't support the microcontroller used on the nano unfortunately, so that part's expected.

I don't think I've actually seen the VM crash before, so that's a first for me as well. 

There's a bit of an oddity with `virtualboxctl` where enabling USB3.0 doesn't disable USB1.0 like it should. In VirtualBox, can you try going to ChipWhisperer Jupyter's settings, select USB, select USB 2.0, hit OK, then select USB 3.0 again?

colinoflynn, 2020-10-01:
Also - if your host OS is Windows, the ChipWhisperer installer might be a quick way to get Python3+ChipWhisperer on Windows. The "installer" just puts everything into a directory so shouldn't leave any mess around if you want to remove it right after.

I didn't realize BOSSA didn't have support for the nano microcontroller - we'll try to build a smaller Python .exe that can do just the updating too.

osok, 2020-10-01:
Tried one more time.  This time I did the following:

1. Unplugged the device for 5 minutes
2. Rebooted the VM
3. Plugged the device in.
4. connected the device to the VM
5. verified it was recognized using `lsusb`, and it was
6. performed a `ls -l /dev/A*` and I saw `/dev/ttyACM0` existed
7. Executed the following:
`import chipwhisperer as cw`
`programmer = cw.SAMFWLoader(scope=None)`
`programmer.program('/dev/ttyACM0', hardware_type='cwnano')`

There was no error and no response.

8. Unplugged for 1 minute
9. Plugged back in
10. device is not recogonized.

I just saw your above post.  I will look in to that now.

colinoflynn, 2020-10-02:
There is now a tool due to @alex-dewar that works on Windows. See the issue #289 for the download (to avoid duplicating).

For this to work, be sure the serial port is not passed through to the VM. Then run:
```
firmware_updater.exe COMN cwnano
```

Where `COMN` should be the com port it's showing up as in Windows.

alex-dewar, 2021-06-17:
Closing this, as there's been no update in a while. If this hasn't been resolved, let me know and I'll reopen

closed 2021-06-17 by alex-dewar

mstoetti, 2022-02-03:
I run in a similar problem where I have to flash manually the CW nano. Is there a SW for linux available or maybe already installed in the CW VM?

alex-dewar, 2022-02-04:
You can erase/flash via the CW Python API: https://chipwhisperer.readthedocs.io/en/latest/firmware.html

mstoetti, 2022-02-04:
HI Alex,

Top thanks! Everything is running!

Best regards,
Marc