newaetech/chipwhisperer #289: ChipWhisperer Firmware Update Tool

colinoflynn, 2020-10-01:
It would be useful to have an update tool that can do the firmware updates, without requiring a "full install". This wouldn't need to support much more than enabling bootloader mode & the serial bootloader.

In theory something like py2exe could let us build a Windows executable for this tool (https://pypi.org/project/py2exe/).

alex-dewar, 2020-10-02:
Py2exe hasn't been update in quite a while and no longer works, but [PyInstaller](https://www.pyinstaller.org/) seems to work fine

colinoflynn, 2020-10-02:
Attached is beta by @alex-dewar 

[firmware_updater_0.01.zip](https://github.com/newaetech/chipwhisperer/files/5319368/firmware_updater_0.01.zip)

Usage:

```
firmware_updater.exe COMPORT CWTYPE
```

The `CWTYPE` is one of `cwlite, cwnano, cw305, cw1200`. Be sure to use correct version.

Example:
```
C:\Users\user\Downloads>firmware_updater.exe COM10 cwlite
Loading cwlite firmware...
Opened!
Connecting...
Connected!
Erasing...
Erased!
Programming file SAM3U_CW1173.bin...
Programmed!
Verifying...
Verify OK!
Bootloader disabled. Please power cycle device.
```

NOTE - this assumes you are ALREADY IN BOOTLOADER MODE. That is the firmware is erased, either with the jumper or with the normal bootloader entry code.

alex-dewar, 2021-05-05:
New versions of the firmware update tool can be found in release 5.5.1

alex-dewar, 2022-07-07:
Closing, as the firmware updater is available from our releases.