newaetech/chipwhisperer #295: Correct path for `cw305_defines.v` in `software/chipwhisperer/capture/targets/CW305.py`

vogelpi, 2020-11-05:
Hi,

we are using the latest version of the ChipWhisperer toolchain from GiHub toolchain in editable mode. When starting the capture using the CW305 and CW-Lite, I get the following error:
```
ERROR:root:Cannot find ['../../hardware/victims/cw305_artixtarget/fpga/common/cw305_defines.v']. Please specify the location of cw305_defines.v on your filesystem.
Traceback (most recent call last):
  File "./simple_capture_traces.py", line 131, in <module>
    ot = initialize_capture(cfg_file['device'], cfg_file['spiflash'])
  File "./simple_capture_traces.py", line 28, in initialize_capture
    ot = device.OpenTitan(
  File "/home/user/ot-sca/cw/cw305/util/device.py", line 18, in __init__
    self.fpga = self.initialize_fpga(bitstream, pll_frequency)
  File "/home/user/ot-sca/cw/cw305/util/device.py", line 27, in initialize_fpga
    fpga.con(bsfile=bitstream, force=False)
  File "/home/user/ot-sca/src/chipwhisperer/software/chipwhisperer/capture/targets/_base.py", line 59, in con
    self._con(scope, **kwargs)
  File "/home/user/ot-sca/src/chipwhisperer/software/chipwhisperer/capture/targets/CW305.py", line 319, in _con
    self.slurp_defines(verilog_defines)
  File "/home/user/ot-sca/src/chipwhisperer/software/chipwhisperer/capture/targets/CW305.py", line 148, in slurp_defines
    defines = open(defines_file, 'r')
FileNotFoundError: [Errno 2] No such file or directory: '../../hardware/victims/cw305_artixtarget/fpga/common/cw305_defines.v'
```

I think the problem is that the path specified in `software/chipwhisperer/capture/targets/CW305.py` is not agnostic to where the API is called from. One should probably first derive the absolute path of the toolchain and then use that to find the defines file.

jpcrypt, 2020-11-05:
Hi Pirmin,
Thank you for pointing this out - I had somehow forgotten about this scenario. You should find this fixed in https://github.com/newaetech/chipwhisperer/commit/9819f42dfa91fd9fc35ff2fb99c509a962a0cf4d.

vogelpi, 2020-11-05:
Thanks a lot @jpcrypt for the quick fix! I can confirm its working now.

I guess you want to leave this issue open until the fix has landed in master. Or shall I close it?

jpcrypt, 2020-11-06:
It can be closed, was just waiting for your confirmation - thanks!

closed 2020-11-06 by jpcrypt