newaetech/chipwhisperer #297: CWLite - Set "trigger" on UART sent (enhancement)

inode-, 2020-11-06:
Hello,

on CWLite the triggers are usually set in the target. But in the real world you have no control over the target, so having the ability to set the trigger right after a UART communication can be very useful. In this way glitches could also be controlled based on UART transmissions making some attacks much easier.

inode

jpcrypt, 2020-11-09:
Some comments on this- the CW-lite FPGA is quite full so it may be not be possible to add more functionality to it.

However, UART sniffing will soon be possible on our PhyWhisperer platform (http://phywhisperer.com/). PhyWhisperer currently has the capability to monitor and trigger on the Arm trace data (https://github.com/newaetech/DesignStartTrace/tree/master/hardware/tracewhisperer); I'm currently adding support to do this via the SWO pin, which is... just UART. So you could use this to monitor UART traffic in general, and trigger with clock-cycle accuracy.

colinoflynn, 2020-11-11:
On the CW-Pro, the UART trigger logic is already there, and *can* run on the CW-Lite FPGA. But as @jpcrypt mentioned, it's basically a fit issue. It's also more complicated than just fitting in available space, but also meeting timing... even very minor additions tend to break timing we've found, the CW-Lite FPGA is more or less a "stable house of cards" in terms of having synthesis meet timing (sometimes need to vary strategies etc).

The PhyWhisperer solution will be the most flexible, as keeps the "hardware" trigger. This means it could trigger on data coming back to the FPGA & have perfectly accurate responses.

If triggering on data sent *from* the ChipWhisperer-Lite, one option is adding a trigger out to the SAM3U. That is - have the SAM3U raise a trigger I/O line when it transmits a byte. This should be perfectly synchronized with everything going on at least.

Note - it's possible to at least trigge ron the falling edge of the start of serial data with the current FPGA. It's far from ideal, but if you have a situation where you want to trigger on the start of a response message it works fairly well. This is shown in https://github.com/newaetech/chipwhisperer-jupyter/blob/master/courses/sca101/Lab%206_4%20-%20Jittery%20Triggering%20on%20UART.ipynb .

inode-, 2020-11-13:
Quite clear everything that you wrote, I know also that the CW-Pro has this kind of feature, but the request is born to be able to use the CW-Lite in a real world scenario and not only for learning. Personally I think that CW is a great tool to learn, but the situation with the targets is too unrealistic. Then when someone tries to apply the concepts on real hardware, problems start to arise.

It is absolutely clear to me that FPGA has limited memory and therefore not all features can be implemented, however I think it is important to be able to review some parts of the code to allow you to use the same hardware even on different targets.

Both in the case of PA and in the case of glitching in many cases we find ourselves in the situation of "sending something" to the target and then starting the analysis from that point.

Having the ability to set the trigger "before sending the data" or "immediately after sending the data" would certainly make CW-Lite more versatile and would also allow you to try to carry out attacks on sample targets without relying on the triggers sent by the target itself making the exercise more realistic.

An alternative is to use another hardware (arduino etc) to manage the UART and set the trigger but in this case I will lose some useful data, like when the chip reset etc.

I didn't understand the possibility proposed by @colinoflynn to have the trigger set by the SAM3U, the serial communication towards the target is not managed directly by the FPGA on the TIO 1-4 ports? Or I missed something?

I will try to use the falling edge on the start UART transmission (from the CW to the target) and using offset reach my goal.

Thank you

colinoflynn, 2020-11-13:
The SAM3U is a minor technical note - it gets around the issue with the FPGA being too full to add new features too.

Basically - the UART trigger *can* work as-is on the CW-Lite, the features can be turned on/off in the FPGA build (the FPGA code is open-source for the Pro, and thus it's not hidden for example). But it won't reliably fit, so you need to disable other features to get this to happen. Unfortunately there is no much to "turn off" - you can easily turn off the glitch logic for example, but presumably if you want to use said glitch logic that isn't so helpful!

But the SAM3U (microcontroller) can be modified. So if the trigger comes from that it would work - the SAM3U is where the 'serial port' is implemented anyway, so for TX there is no sync issues (since the trigger would be lock-step with the clock of the SAM3U). This trigger would in fact be routed to the FPGA trigger input (i.e., inside the FPGA it looks like this came from some external source).

The medium-term goal is an updated CW-Lite with a larger FPGA to put stuff like this into it, which is where most effort at this point will end up. The PhyWhisperer-USB is being used to build new features here, so we have a bit of a "technical platform" to develop them (as J-P mentioned, it's a new architecture which *greatly* improves the triggering, which is the other reason we have been concentrating on that).

eychei, 2021-06-08:
Hello,

is there an update on this matter? 
I did see the new firmware that does have USB CDC for the CWLite. Is thos somehow connected to this topic?
Also some Tutorials mention triggering on the Rx or Tx line (IO4) with the CWLite.
Hope someone can give an update. 

-e

alex-dewar, 2021-06-08:
The USB CDC isn't connected to this - it just adds a OS serial port (`/dev/tty*` or `COMX`) that can be used along with the typical target communication. 

The FPGA can already trigger off the serial_tx or serial_rx lines, which is done in `sca101/Lab 6_4`. This is the same as triggering on other pins (i.e. a rising or falling edge trigger) - there's no UART decoding capabilities on the FPGA due to the aforementioned FPGA space limitations.  The serial is all done on the SAM3U, so it should be possible to set a pin on the SAM3U high once the serial message is sent, then trigger the FPGA on that.

eychei, 2021-06-08:
So the USB CDC is just for "watching" the target response and maybe use this in other software simultaneously?

Looking at sca101/Lab 6_4 i can see that we can use tx and rx as triggers. What benefit would we have in using an external trigger (IO4) in this specific case?

Also I did read the the IO pins are not 5V tolerant for the CWLite. So using the trigger of my Rigol DS1052e would most likely fry the board right? 

-e

alex-dewar, 2021-06-08:
Yup, CDC accomplishes the same thing that the normal `cw.target()` SimpleSerial stuff does.

The advantage of a dedicated trigger is proximity to the operation you care about (less time between the trigger and the encryption, for example), as well as less jitter. If you do the lab, you'll see that the traces you capture triggering off the TX pin are a bit jittery. This can be fixed by post processing.

Yes, the IO pins are 3.3V on the Lite. You'll need to use a level shifter if you want to use the trigger from your scope to trigger the chipwhisperer.

eychei, 2021-06-08:
Thanks for the clarification.
Great support and response time.

-e