newaetech/chipwhisperer #300: Programming Firmware on target CW303_XMEGA failed

ben09310, 2020-12-30:
hello all, 

my code:

SCOPETYPE = 'OPENADC'
PLATFORM = 'CWLITEXMEGA'
SS_VER = 'SS_VER_1_1'
%run "../../Setup_Scripts/Setup_Generic.ipynb"
%%bash -s "$PLATFORM" "$SS_VER"
cd ../../../hardware/victims/firmware/basic-passwdcheck
make PLATFORM=$1 CRYPTO_TARGET=NONE SS_VER=$2
fw_path = "chipwhisperer/hardware/victims/firmware/bootloader-aes256/bootloader-aes256-{}.hex".format(PLATFORM)
cw.program_target(scope, prog, fw_path)

error:

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-6-b876eeece5c0> in <module>
----> 1 cw.program_target(scope, prog, fw_path)

~/Videos/chipwhisperer/software/chipwhisperer/__init__.py in program_target(scope, prog_type, fw_path, **kwargs)
     60     prog._logging = None
     61     prog.open()
---> 62     prog.find()
     63     prog.erase()
     64     prog.program(fw_path, memtype="flash", verify=True)

~/Videos/chipwhisperer/software/chipwhisperer/capture/api/programmers.py in func_wrapper(self, *args, **kwargs)
     68             self.scope.io.nrst = 'high_z'
     69         try:
---> 70             val = func(self, *args, **kwargs)
     71         finally:
     72             logging.debug('Restoring pdic, pdid, and nrst pin configuration')

~/Videos/chipwhisperer/software/chipwhisperer/capture/api/programmers.py in find(self, xmega)
    223     def find(self, xmega=None):
    224         xmega = self.xmegaprog()
--> 225         sig, chip = xmega.find()
    226
    227         # Print signature of unknown device

~/Videos/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py in find(self)
    173     def find(self):
    174         self.setParamTimeout(100)
--> 175         self.enablePDI(True)
    176
    177         # Read signature bytes

~/Videos/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py in enablePDI(self, status)
    307         if status:
    308             # self._xmegaDoWrite(self.XPROG_CMD_LEAVE_PROGMODE)
--> 309             self._xmegaDoWrite(self.XPROG_CMD_ENTER_PROGMODE)
    310             self._pdienabled = True
    311         else:

~/Videos/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py in _xmegaDoWrite(self, cmd, data, checkStatus)
    502             status = self._xmegaDoRead(cmd=0x0020, dlen=3)
    503             if status[1] != 0x00:
--> 504                 raise IOError("XMEGA Command %x failed: err=%x, timeout=%d" % (status[0], status[1], status[2]))
    505
    506     def _xmegaDoRead(self, cmd, dlen=1):

OSError: XMEGA Command 20 failed: err=1, timeout=1

Have anybody get the same error code?

ben

colinoflynn, 2020-12-30:
What's the hardware setup? Is this on a ChipWhisperer-Lite XMEGA board (i.e., 
[image] ) or something else with an XMEGA attached?

The error **_OSError: XMEGA Command 20 failed: err=1, timeout=1_** is that it didn't find the attached XMEGA device. The "easiest" solution is you're using a different hardware (ChipWhisperer-Lite ARM looks very similar).

However it could also be if you have something on the IO lines interfering with programming, so more details of the physical HW are needed.

ben09310, 2020-12-30:
yes, i have a  ChipWhisperer-Lite XMEGA board. 
when i change the PLATFORM = 'CWLITEXMEGA' to PLATFORM = 'CWLITEARM'
and run the code i get the error code 

OSError: Could not detect STM32F, check connections, BOOT MODE entry setup

i'm sure the target board is XMEGA and is conntected, because i get no error code when i run 
scope = cw.scope()

what details to you need?

i'm a beginner and i want to test the hardware with the program "PA_Intro_1-Firmware_Build_Setup.ipynb"
path "chipwhisperer/jupyter/archive"

alex-dewar, 2023-10-13:
Closing, as I believe this was resolved through support@newae.com

closed 2023-10-13 by alex-dewar