newaetech/chipwhisperer #305: Some issues with CWnano

uranusb, 2021-02-05:
Hi,

I am a beginner in ChipWhisperer. I got a CWnano recently and have installed the VM. After connecting the device with the VM, I tried to run some example codes to check if the board works fine. 

Firstly I run following codes:
**import chipwhisperer as cw
scope = cw.scope()
target = cw.target(scope, cw.targets.SimpleSerial)**
and got some output as follows:
**Serial baud rate = 38400**

Then I tested
**%%bash
cd ../hardware/victims/firmware/simpleserial-base/
make PLATFORM=CWNANO CRYPTO_TARGET=NONE**
and everything also worked fine.

However, when I tried following codes:
**import chipwhisperer as cw
scope = cw.scope(type=scopes.CWNano)**
it showed an error as follows:
**NameError: name 'scopes' is not defined**

Moreover, I also tested some basic commands as follows:
**msg = bytearray([0]\*16) #simpleserial uses bytearrays
target.simpleserial_write('p', msg)
print(target.simpleserial_read('r', 16))**
And also got a warning massage 
**WARNING:root:Unexpected start to command: (some garbled message)**

Can you show me how to solve this problem? Any help will be appreciated

Thanks

alex-dewar, 2021-02-10:
Hi,

If you're already connected, you shouldn't try to connect by running `cw.scope()` again (also it should be `cw.scopes.CWNano`). It also doesn't look like you're programming the target in the example above (should be `cw.program_target(...)`), which means the target may not be running any code and therefore won't respond correctly to messages.

alex-dewar, 2021-04-06:
Hi @uranusb 

Are you still having this issue?

uranusb, 2021-04-07:
> Hi @uranusb
> 
> Are you still having this issue?

The Problem has been solved, thanks!

alex-dewar, 2021-04-07:
Good to hear!

closed 2021-04-07 by alex-dewar