newaetech/chipwhisperer #306: get_fpga_buildtime doesn't work for CW305, always returns "FPGA build time: 0/0/2000, 0:0"

alphan, 2021-02-13:
Ran into this while trying to figure out if the FPGA on CW305 needs to be programmed or not.

Reproduction steps:
```
>>> import chipwhisperer as cw
>>> fpga = cw.capture.targets.CW305()
>>> fpga.con() # Return value is the same whether the FPGA is programmed or not.
>>> print(fpga.get_fpga_buildtime())
FPGA build time: 0/0/2000, 0:0
```

jpcrypt, 2021-02-13:
Hi Alphan,

You need to enable the timestamp feature in your Vivado project.
See https://www.xilinx.com/support/documentation/application_notes/xapp1232-bitstream-id-with-usr_access.pdf, under "Vivado Tools Flow" on page 3.

Jean-Pierre

alphan, 2021-02-13:
Thanks @jpcrypt!

alphan, 2021-02-25:
Hi @jpcrypt, 

We generated a new bitstream w/ the USR_ACCESS TIMESTAMP feature (please see lowRISC/opentitan#5279) but `fpga.get_fpga_buildtime()` still returns `FPGA build time: 0/0/2000, 0:0`. Would you mind taking a look at that PR to see if we missed anything before we debug further? Thanks!

jpcrypt, 2021-02-25:
Hi Alphan,
That date is what you get when the register read returns all zeros.
Your PR looks correct in that the .tcl script is producing the correct command.
I would check whether it actually gets applied when running the implementation.
In my case, the TIMESTAMP command is in my .xdc constraint file which is used for both synthesis and implementation, and in the implementation runme.log logfile, I see the following, near the very end:
```
(...)
INFO: [Designutils 20-2272] Running write_bitstream with 2 threads.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Overwriting "TIMESTAMP" with "B92AFA2E" for option USR_ACCESS
TIMESTAMP = Tue Feb 23 15:40:46 2021

Creating bitmap...
Creating bitstream...
Writing bitstream ./cwhusky_top.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Common 17-83] Releasing license: Implementation
152 Infos, 13 Warnings, 3 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:28 ; elapsed = 00:00:21 . Memory (MB): peak = 2560.469 ; gain = 425.609
INFO: [Common 17-206] Exiting Vivado at Tue Feb 23 15:40:55 2021...
```

jpcrypt, 2021-02-25:
@alphan it strikes me just now that you may not be using our standard SAM3U <-> FPGA register read/write interface in your own design? Because that's what `get_fpga_buildtime()` uses to read that buildtime value. If that's the case, have a look at our example CW305 designs, either the AES one or the ECC one, in https://github.com/newaetech/chipwhisperer/tree/develop/hardware/victims/cw305_artixtarget/fpga/vivado_examples

The buildtime register is here: https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/cw305_artixtarget/fpga/common/cw305_reg_aes.v

It's just a matter then of returning that 4-byte timestamp value through whatever interface your host uses to communicate with the FPGA.