newaetech/chipwhisperer #307: Cannot tell if the FPGA on CW305 is programmed by con

alphan, 2021-02-13:
For CW305, it makes sense to call `con` with `force=False` to save time but it would be nice to have a way to tell if the fpga was actually programmed or not, e.g. for more accurate messages. Since `con` both connects to and programs the fpga, clients cannot get this information before calling `con` and the FPGA is already programmed after calling `con`, either because it was already programmed before calling `con` or because it was programmed in `con`.

jpcrypt, 2021-02-14:
Hi Alphan,

I'm not sure if I understand what you're looking for, but what if you did something like this:

```Python
target = cw.target(scope, bsfile=bs, force=False)
if not target.fpga.isFPGAProgrammed():
    status = target.fpga.FPGAProgram(open(bs, 'rb'))
    if status:
        print('FPGA programmed, buildtime = %s' % target.get_fpga_buildtime())
    else:
        print('FPGA programming failed')
```

alex-dewar, 2023-10-24:
Should be implemented by https://github.com/newaetech/chipwhisperer/commit/8a0bc31071f75366cef6dee2207fc2dcf83be747