newaetech/chipwhisperer #311: PicoScope, Sakura-G and Sakura-W Integration

Rodrigojflima, 2021-03-01:
Hello guys,  I'm working (for my msc. thesis) on the development of a platform to evaluate the resistance of devices against side-channel attacks.

I'm using the ChipWhisperer project software due to the great features it already provides. Regarding equipment, I own a PicoScope 6404D and 2 SAKURA boards (G and W). I'm aware that the capture and analysis modules are quite separate in CW5, however, ideally, I would like to integrate this equipment to work with the ChipWhisperer software seamlessly (making it easier for others to work with it as well).

Will I be capable of making the PicoScope to be a cw scope object (essentially making use of the pico-python library) and the SAKURA boards to be target objects? If so, since the organization of this project is still quite confusing to me, so, are there any guidelines that could be provided to facilitate my work?

colinoflynn, 2021-03-01:
You might find some people on forum.newae.com that have tried some of this before. We used to have external (picoscope-based) support but it got outdated so we removed the module. Generally adding large new modules directly is tricky, as the APIs tend to change over time and we don't want to support these external modules.

The current CW design tries to make it easier to use portions of the code without needing to "fully integrate" new modules. For example there is a module to generate ciphertext + plaintext, analyze, etc. You can feed those into your target device & get the responses back.

We do have a SAKURA-G module, but IIRC it uses an outdated bitstream. There is a post on the forum with the required information. This would at least let you directly control the SAKURA-G target board from ChipWhisperer for example.

At least initially you'll need to get the Python comms up for your device & scope. For the scope for example you can easily save trace files into a ChipWhisperer project file, without actually needing to "fully integrate" the scope.

Rodrigojflima, 2021-03-03:
Thank you for answering Colin. I've tried to find the post containing the required information that you mentioned but was unsuccessful. Are you able to share the link to it here?

alex-dewar, 2021-03-03:
Just want to mention that we do have an alpha port of the SAKURA-G into CW5 (not sure if that was what Colin was talking about there). IIRC connecting should be something like:

```python
import chipwhisperer as cw
scope = cw.scope(cw.scopes.SakuraG)
target = cw.target(scope, cw.targets.SakuraG)
```

Rodrigojflima, 2021-03-04:
This prepared module implies the use of the OpenADC hardware, am I right? The intention would be to allow the PicoScope to be a scope object and the SAKURA-G to be the target object.

jmichelp, 2021-03-04:
Hi @Rodrigojflima,

I'm actually working on bringing back PicoScope to CW5 software stack, also for a PS6404D unit.

I have working code but unfortunately the data buffer that I get back is corrupted: if I use the official GUI I have a clean trace, doing the same programmatically gives garbage (even when I connect the input channel to the calibration port!).

How much in a rush are you for this? I'm hoping to have fixed the bug by the end of the month. That's a pretty tricky one to identify.

Rodrigojflima, 2021-03-05:
Hi @jmichelp,

That's actually pretty good news! I'm currently working on developing my own SAKURA-G target module so I not only can better understand the global functioning of this project but also get it to be as I would need. That to say that I'm not in much rush. Take your time, but please get back to me, I'm very interested in taking a look at that!

colinoflynn, 2021-03-05:
On the SAKURA-G - check https://forum.newae.com/t/sakura-g-target-example/45/6 for example, but if you search forum.newae.com there are more hits.

As @jmichelp mentions about "Bringing back" - the old CW4 version was written in Python2.7 so some features were dropped that used to exist. Picoscope support was there along with SAKURA-G as a target + a scope. Using the SAKURA-G as a "scope" mean with an on-board OpenADC module, but at the same time running code in the target FPGA side.

colinoflynn, 2021-03-05:
PS - on the SAKURA-G by the "dual use" - you can enumerate it as a scope to get the com interface up, then ignore the scope object. I think you need ot pass it to the "target" interface. It should "just work" with the firmware posted on the forum IIRC.

Rodrigojflima, 2021-03-15:
I've already developed my new and updated target object for the SAKURA-G. I'm now working on the scope object for the PicoScope 6404D. Once things are working as they should, I will send a pull request to be evaluated by you guys, and if it contributes in any way, I'm more than happy to see it being merged. 

But I'm actually coming by now to ask @jmichelp if there is any way that he could share the code he has so far for the PicoScope,  as it will be helpful for me, and since I'm working on that now, I might be able to help a bit as well.

closed 2021-03-25 by Rodrigojflima

mattiamirigaldi, 2023-10-18:
Hi @jmichelp @Rodrigojflima 
I'm also working on getting the piscoscope to work as a cw scope. Have you been able to make it work? If so, would you mind linking to your project? 
My setup is the CW305 Artix 100 as target and the picoscope 5000 as scope.