newaetech/chipwhisperer #316: udev rules possibly faulty, error in journalctl -b

bigla, 2021-03-10:
The udev-rules shown in https://chipwhisperer.readthedocs.io/en/latest/prerequisites.html#hardware-drivers
yields the following output in `journalctl -b`:
```
systemd-udevd[252]: /etc/udev/rules.d/99-newae.rules:5 Unknown group 'plugdev', ignoring
systemd-udevd[252]: /etc/udev/rules.d/99-newae.rules:8 Unknown group 'plugdev', ignoring
systemd-udevd[252]: /etc/udev/rules.d/99-newae.rules:11 Unknown group 'plugdev', ignoring
systemd-udevd[252]: /etc/udev/rules.d/99-newae.rules:12 Unknown group 'plugdev', ignoring
```
Comment 2 in this [bug report](https://bugs.archlinux.org/task/47205) leads to a possible solution.

> You can use TAG+="uaccess" to grant at-seat permissions to the device

colinoflynn, 2021-03-10:
Thanks - yeah we reused the `plugdev` solution from since forever, I didn't realize there was a "nice" solution like this - thanks!

alex-dewar, 2021-03-12:
Thanks for the report! Fixed by 5684f1d43ee5eea3109bc8c4d91ef2ffe045b1a3

closed 2021-03-12 by alex-dewar