newaetech/chipwhisperer #323: how to set new key

Aegon007, 2021-03-31:
Hi,

I am new to chipwhisperer, I was trying to generate the key-text pair with the code"
"
ktp = cw.ktp.Basic()
key, text = ktp.next()
"
and it give me the key:  b'2b 7e 15 16 28 ae d2 a6 ab f7 15 88 09 cf 4f 3c'

however, after I use the code below:
"
new_key = b'3b 74 1a 16 28 af d2 e6 ab f7 25 78 09 cf 3f 2c'
target.set_key(new_key)
key, text = ktp.next()
"
it did not return the new key, it still give me the old key.

so my question is: how to set new key?

alex-dewar, 2021-03-31:
The key is just a bytearray, so if you want to use a different key, I'd recommend just doing `key = bytearray([0x3b, 0x74, ..., 0x2c])`.

alex-dewar, 2021-04-06:
Closing, as I believe the above answers your question.

closed 2021-04-06 by alex-dewar