newaetech/chipwhisperer #325: CW305 vivado aes problem

sonersec, 2021-04-06:
Vivado 2020.2 gives error while compiling Aes128 for CW305.

jpcrypt, 2021-04-06:
What are the errors? Are they related to missing ILAs?

sonersec, 2021-04-06:
> What are the errors? Are they related to missing ILAs?

Yes. It says ila_0 and ila_1 are missing.

jpcrypt, 2021-04-06:
Right; you can either undefine ILA_REG and ILA_CRYPTO, or generate yourself the IP for the missing ILAs from Vivado.
I've done the former in https://github.com/newaetech/chipwhisperer/commit/c2503abd53c872998786c604aff79f013c456cab.

closed 2021-04-06 by jpcrypt

sonersec, 2021-04-07:
The fix does not work. Vivado still does not sytnhetise. But it did not give any error. Just waiting to do synthitise. What are other options to build it?

sonersec, 2021-04-07:
@jpcrypt The issue continues as mentioned above.

jpcrypt, 2021-04-07:
I really can't help you without additional information. Help me by answering the following questions:
1- Do you have any experience using Vivado?
2- Please describe exactly what you've done.
3- What does "Vivado doesn't synthesize" mean: what exactly happens when you click on "run synthesis"?

sonersec, 2021-04-07:
@jpcrypt 
1- I have little experience.
2- I downloaded the project and open the project in Vivado 2020.2
3- When I run synthesis, it starts cycling a long time and never finishes synthesis.

q1) Do you try to synthesize the project after the modifications?

jpcrypt, 2021-04-07:
What's a "long time"? This is not like compiling software, Vivado is not fast. It can take around 5-10 minutes.

I did run synthesis on the latest version, but I have Vivado 2019.1. I will check with 2020.2 when I have the chance, but I don't have enough diskspace to install that right now. I doubt this would be the issue.

Are you aware that the chipwhisperer releases include pre-compiled FPGA bitfiles? There is no need to generate your own, unless you intend to modify the Verilog source code.

sonersec, 2021-04-07:
@jpcrypt 
Long time means I leave it one day.
I need to modify the verilog code because I am doing an academic research.

colinoflynn, 2021-04-07:
On 2020.2 with a fresh check-out, you'll see this warning but you can ignore it:
[image]

Pressing the "Generic Bitstream" button:
[image]

You'll see this message come up:
[image]

Does it change at all? After a few minute you get this status for example:
[image]

Then that will finish etc. I tested by changing the file `aes_sbox.v` as follows:
[image]

I can confirm it works the same. There is a moment at the start where it will say "Queued" on the status window but it changes after about 30 seconds to "Running synth_design". Otherwise it sounds like a Vivado install problem - you might try checking with a basic project it works? I know Vivado can have issues if antivirus is running on Windows (antivirus once kept *removing* my binaries from Vivado which took me a bit to figure out what was happening). You can install Vivado in a VM as well on Linux but it's slow I warn you...

sonersec, 2021-04-07:
I closed the antivirus. It passed synthetise but it still failed in the implementation.

[image]

sonersec, 2021-04-07:
I closed the antivirus and it works. Thanks for the help. I never think the antivirus is the problem.

colinoflynn, 2021-04-07:
LOL ok crazy... glad to know I'm not the first person Vivado + Anti-Virus has caused problems for. Glad you got it working - not sure on long-term solution, I had to just exempt my Vivado binary from antivirus scanning...