newaetech/chipwhisperer #328: CW1200 - nRST Capture trigger not working

opcod3, 2021-04-12:
Attempting to capture a trace with the CW1200 using nRST as trigger does produce any output, the capture simply times out.
If a jumper is placed between nRST and GPIO4 and TIO4 is set as trigger the trace is captured correctly.

While I am almost completely ignorant in HDL, looking at commit fb0ba7a22f9 it appears the interface code was only updated for the CWLite and not the CW1200.

colinoflynn, 2021-04-12:
We almost got it....

This was a known issue we discovered last week & did a new FPGA build to fix that (it was on the build server so not pushed) - give me a few mins and I'll give you a development bitstream to test with.

colinoflynn, 2021-04-12:
Actually it looks to be fixed at https://github.com/newaetech/chipwhisperer/commit/49260f902e2018c931d3ad08f15c8b42092649fc but not rebuilt.

I've just pushed a commit with the rebuilt (I think) fix. Untested yet but will get too that shortly... in the meantime see if it "just works" now.

opcod3, 2021-04-12:
Alright, I'll try to check it out. Thanks for the quick reply!

closed 2021-04-30 by alex-dewar