newaetech/chipwhisperer #330: PCB files for CWnano

anandvsalunke, 2021-04-15:
I was looking forward to get started with CW nano, would like to build this myself.
looking for PCB layout files

colinoflynn, 2021-04-15:
The CWNano wasn't fully OSHW like the CW-Lite with all documentation, although we did end up releasing schematic etc so it's been remixed into other boards too. We didn't do PCB files for our version on account of support issues (not individual people building it - but if it ends up on aliexpress and we get a bunch of emails with problems). Note sure where you are in the world but if we've got blank from previous runs can send you a few blank ones if it works (mostly - if mail won't take months to get to you!).

closed 2021-04-15 by colinoflynn

anandvsalunke, 2021-04-15:
Thanks for your prompt response. I am from India, it would be great if you can send a blank PCB.
Following the schematic, it would be easy to design PCB from scratch, it would also help if you can help with hi res photo of PCB (Front & back), layer stack-up information & BOM.