newaetech/chipwhisperer #334: Trace manager parses config differently than the project class

Phasip, 2021-04-29:
```
>>> import chipwhisperer as cw
>>> p = cw.create_project('demo')
>>> p.setAuthor("Me\nMyself")
>>> p.traces.append(cw.Trace([0],None,None,None))
>>> p.save()
>>> p2 = cw.open_project('demo')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/develop/ChipWhisperer/chipwhisperer/software/chipwhisperer/__init__.py", line 96, in open_project
    proj.load(filename)
  File "/home/develop/ChipWhisperer/chipwhisperer/software/chipwhisperer/common/api/ProjectFormat.py", line 302, in load
    self._traceManager.loadProject(self.filename)
  File "/home/develop/ChipWhisperer/chipwhisperer/software/chipwhisperer/common/api/TraceManager.py", line 89, in loadProject
    config.read(configfilename)
  File "/usr/lib/python3.8/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/usr/lib/python3.8/configparser.py", line 1113, in _read
    raise e
configparser.ParsingError: Source contains parsing errors: '/home/develop/Desktop/demo.cwp'
	[line  9]: "Myself'''\n"
>>> # How it is loaded by the Project class
>>> c = cw.common.api.ProjectFormat.ConfigObjProj(infile="/home/develop/Desktop/demo.cwp")
>>> c
ConfigObj({'Trace Management': {'tracefile0': 'demo_data/traces/config_2021.04.28-23.29.23_0.cfg', 'enabled0': 'True'}, 'ChipWhisperer': {'General Settings': {'Project Name': 'Untitled', 'Project File Version': '1.00', 'Project Author': 'Me\nMyself', 'Program Name': 'ChipWhisperer', 'Program Version': ''}}})
```

The simple fix seems to ensure that the tracemanager uses `ConfigObjProj` to read the config instead of `configparser`

mcsmonk, 2021-04-29:
Each method use configparser [pypi](https://pypi.org/project/configparser/) and configobj [pypi](https://pypi.org/project/ConfigObject/), respectively.

Since each package treats the config file in a different way, it seems that you experienced it.

refer to [https://stackoverflow.com/questions/34130626/python-configparser-escape-newlines](https://stackoverflow.com/questions/34130626/python-configparser-escape-newlines)

mcsmonk, 2021-04-29:
Each method use two different packages, configparser \[[pypi](https://pypi.org/project/configparser/)\] and configobj \[[pypi](https://pypi.org/project/ConfigObject/)\], respectively. Since each package treats the config file in a different way, it seems that you experienced it.
refer to [https://stackoverflow.com/questions/34130626/python-configparser-escape-newlines](https://stackoverflow.com/questions/34130626/python-configparser-escape-newlines) and [https://stackoverflow.com/questions/11399665/new-lines-with-configparser](https://stackoverflow.com/questions/11399665/new-lines-with-configparser).

alex-dewar, 2023-10-13:
We hope to replace the project stuff at some point, so we probably won't do any non-essential fixes for project stuff. This also isn't really part of our API, so am closing this issue.

closed 2023-10-13 by alex-dewar