newaetech/chipwhisperer #336: SimpleSerial `WARNING:root:Unexpected start to command:`

viniul, 2021-05-01:
I have a CWLite with an `STM32F4` target. Using the chipwhisperer virtual machine (version 5.5.1) I get a `WARNING:root:Unexpected start to command:  None` when I try out a the `simpleserial-base` example. How would I go about debugging this?

I executed the following commands in the `jupyer` notebook:
<pre>
import chipwhisperer as cw
scope = cw.scope()
target = cw.target(scope, cw.targets.SimpleSerial)
scope.default_setup()
cw.program_target(scope, cw.programmers.STM32FProgrammer, "../hardware/victims/firmware/simpleserial-base/simpleserial-base-CW308_STM32F4.hex")
</pre>
This seems to work:
<pre>
Serial baud rate = 115200
Detected known STMF32: STM32F40xxx/41xxx
Extended erase (0x44), this can take ten seconds or more
Attempting to program 5315 bytes at 0x8000000
STM32F Programming flash...
STM32F Reading flash...
Verified flash OK, 5315 bytes
Serial baud rate = 38400
</pre>
Now I build the `simpleserial-base` example and want to send a simple message:
</pre>
%%bash
cd ../hardware/victims/firmware/simpleserial-base/
make PLATFORM="CW308_STM32F4" CRYPTO_TARGET=NONE
msg = bytearray([1,2,3,4])
target.simpleserial_write("p",msg)
print(target.simpleserial_read("r", 1000, timeout=10000))
</pre>
which returns the error:
<pre>
WARNING:root:Unexpected start to command: 
None
</pre>
Printing the variables `scope` and `target`:
<pre>
print(scope)
cwlite Device
sn         = 50203220313038543230323239313037
fw_version = 
    major = 0
    minor = 30
    debug = 0
gain = 
    mode = low
    gain = 0
    db   = 5.5
adc = 
    state          = False
    basic_mode     = low
    timeout        = 2
    offset         = 0
    presamples     = 0
    samples        = 0
    decimate       = 1
    trig_count     = 6010355
    fifo_fill_mode = normal
clock = 
    adc_src       = clkgen_x1
    adc_phase     = 0
    adc_freq      = 0
    adc_rate      = 0.0
    adc_locked    = True
    freq_ctr      = 0
    freq_ctr_src  = extclk
    clkgen_src    = system
    extclk_freq   = 10000000
    clkgen_mul    = 2
    clkgen_div    = 1
    clkgen_freq   = 192000000.0
    clkgen_locked = True
trigger = 
    triggers = tio4
    module   = basic
io = 
    tio1         = serial_tx
    tio2         = serial_rx
    tio3         = high_z
    tio4         = high_z
    pdid         = high_z
    pdic         = high_z
    nrst         = high_z
    glitch_hp    = False
    glitch_lp    = False
    extclk_src   = hs1
    hs2          = None
    target_pwr   = True
    tio_states   = (1, 1, 0, 0)
    cdc_settings = array('B', [1, 1])
glitch = 
    clk_src     = target
    width       = 10.15625
    width_fine  = 0
    offset      = 10.15625
    offset_fine = 0
    trigger_src = manual
    arm_timing  = after_scope
    ext_offset  = 0
    repeat      = 1
    output      = clock_xor
   </pre>
<pre>
print(target)
Serial baud rate = 38400
SimpleSerial Settings =
	output_len             = 16
	baud                   = 38400
	simpleserial_last_read = 
	simpleserial_last_sent = 
	

</pre>

mcsmonk, 2021-05-02:
```
%%bash
cd ../hardware/victims/firmware/simpleserial-base/
make PLATFORM="CW308_STM32F4" CRYPTO_TARGET=NONE
msg = bytearray([1,2,3,4])
target.simpleserial_write("p",msg)
print(target.simpleserial_read("r", 1000, timeout=10000))
```

The above code must be executed separately in different cells as follows.

cell for bash
```
%%bash
cd ../hardware/victims/firmware/simpleserial-base/
make PLATFORM="CW308_STM32F4" CRYPTO_TARGET=NONE
```
cell for python
```
msg = bytearray([1,2,3,4])
target.simpleserial_write("p",msg)
print(target.simpleserial_read("r", 1000, timeout=10000))
```

refer to [https://ipython.readthedocs.io/en/stable/interactive/magics.html#cell-magics](https://ipython.readthedocs.io/en/stable/interactive/magics.html#cell-magics)

viniul, 2021-05-02:
Thanks for your input @mcsmonk. I reordered as you described (the cells were separated in the first place, I should have made that clear, sorry!). Unfortunately, I still get the same error. PFA to this issue a zip contained the respective notebook in case you are interested.
[cw_viniul.zip](https://github.com/newaetech/chipwhisperer/files/6412036/cw_viniul.zip)

mcsmonk, 2021-05-03:
In cw_viniul.zip, the below code is wrong

```python
target.simpleserial_write("p",msg)
print(target.simpleserial_read("r", 1000, timeout=10000))
```

See the **simpleserial-base.c**. The code return 16byte as follows.

```python
simpleserial_put('r', 16, pt);
```

You must read same byte as follows.

```python
target.simpleserial_write("p",msg)
print(target.simpleserial_read("r", 16, timeout=10000))
```

viniul, 2021-05-03:
Thanks @mcsmonk. I changed the code as you suggested, unfortunately I still get the same error message.

mcsmonk, 2021-05-03:
Not only length of output, but also length of input must be same.
See the [line](https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/firmware/simpleserial-aes/simpleserial-aes.c#L161)
```python
# simpleserial-aes.c
simpleserial_addcmd('p', 16,  get_pt);
```

But, you input only 4 bytes.
```
# cw_viniul.ipynb
#msg = bytearray([1,2,3,4])
target.simpleserial_write("p",msg)
```

Try message input of 16 bytes.

viniul, 2021-05-03:
That did it, thanks!

closed 2021-05-03 by viniul