newaetech/chipwhisperer #340: ChipWhisperer clock instability with newer FPGA build

colinoflynn, 2021-05-24:
(Reported on Discord - copying here)

Reported issue with clock output being at 7MHz instead of 7.37MHz.

Using old bitstream by copying https://github.com/newaetech/chipwhisperer/blob/9c8b5560fb6a954d24ee97dc4eb3091ff0a46484/software/chipwhisperer/hardware/firmware/cwlite.py in place & restarting Jupyter kernel + then unplugging/replugging ChipWhisperer seemed to fix it.

Need to test the CW-Lite to confirm no issues with clock locking on latest build. Possibly add check of output frequency to confirm it's as expected to detect this for now.

alex-dewar, 2022-07-07:
A new fpga bitstream has been released with CW 5.6.1, but, as this issue is difficult to replicate, we're not sure if this is fixed

alex-dewar, 2023-10-13:
Don't think we've gotten any complaints about this in a while, so this is probably fixed.

closed 2023-10-13 by alex-dewar