newaetech/chipwhisperer #341: Faulty Trace Captures

qantik, 2021-06-08:
I noticed that our CW308 UFO board started to record incorrect traces after a capturing a certain
number of correct samples, meaning that the signal becomes static noise after a while.

In the screenshot below, we run `TINYAES128C` from the `simpleserial-aes` tutorial on the
`STMF0` target board with `scope.adc.samples = 20000`. The first three round functions are clearly observable
whereas the samples after 17500 become strangely noisy.

This is a new issue I haven't encountered before. The same thing happens for
other targets, i.e., `STMF1`, `STMF3`, `XMEGA` at different time points.
It is also independent of the actual code that is running.

I'm wondering whether our ChipWhisperer board is broken?

[image]

colinoflynn, 2021-06-08:
Hmm - so this might be related to another issue we've been chasing. 

Are you using a newer software release? Issue #340 flags another FPGA issue, what you are showing looks a little like some clocking issues. Can you revert that one file to check if that changes what you see?

qantik, 2021-06-08:
I'm using commit `d7a0930d8ed51e078544a50658e0c8893afff022`. I replaced the file as indicated in #340
but the issue persists.

colinoflynn, 2021-06-08:
Hmm, you're capturing w/ a ChipWhisperer-Lite board right, like stand-alone capture? We could swap that board to confirm it's not the main unit HW if quicker to debug. If so drop an email to sales@newae.com & include - note about CW-Lite capture, shipping address, shipping phone # (this is required by the courier - won't be connected to anything else).

I'm not sure why it would seem to work at the start & fail later based just on HW. But I haven't seen that issue before and I think we would have run into it if repeatable as well, so a HW-Swap might be faster.

colinoflynn, 2021-06-08:
Another comment I had - was if this was running code that aborts the AES early. When we build some of the examples we would do that (for export reasons - so it won't run actual AES).  But does it sometimes work and sometimes not? Like it's clearly an analog problem?

qantik, 2021-06-08:
Yes, it's the ChipWhisperer-Lite board. I will try to dig a little deeper. I keep you posted.

alex-dewar, 2021-06-09:
> 
> 
> I'm using commit `d7a0930d8ed51e078544a50658e0c8893afff022`. I replaced the file as indicated in #340
> but the issue persists.

Just to make sure, have you unplugged + replugged the ChipWhisperer + restarted Jupyter after replacing the file? If not, you
probably still on the old FPGA firmware.

alex-dewar, 2023-10-13:
Will be closing this soon, as there's been no update in a while.

closed 2023-10-18 by alex-dewar