newaetech/chipwhisperer #345: AVR Programmer does not change lock bit of Notduino(CW304)

fallazc, 2021-06-20:
I have been trying to change the lock bits for this target but could not get it to work.
Here is what I tried.
```python
%run "../../../jupyter/Setup_Scripts/Setup_Notduino.ipynb"
```
```python
avr = scope.scopetype.dev.avr
avr.find()
avr.eraseChip()
avr.writeLock(0xFF)           # No error
print(avr.readLock(True))   # Still return 0
```
At first, sometimes, it will return the wrong value (8) but now it just returns 0.
Given the fact that I can change the flash a new firmware and the following flash verification succeeds, the value returned by avr.readLock() is obviously incorrect.

Any ideas what might be the issue here?

colinoflynn, 2021-06-21:
Hmm - we do use the lock bit writing in previous versions so it "should" work - does find() print the correct chip detected?

We used to need to use a "slow clock mode" to slow down the SPI communication when the device had default fuse bytes, as it would have the "divide by 8" option. If your AVR has default fuse bytes (aka - a brand new chip) I suspect you're running into this. I'll have to check what this API is.

alex-dewar, 2021-06-21:
As with the XMEGA, the ATMEGA328p doesn't allow you to set lock bits from 0 to 1 - the only way to set them back from 0 to 1 is to use the chip erase command. 

If you're seeing the lock bits read as zero, there's probably something wrong with the lock bit read. I'll take a quick look and see if I can find anything.

fallazc, 2021-06-21:
Yes that is what I was thinking too. I tried to read the values directly from the board and send it via serial and get `0xFF` which makes sense. And yes, find() returns the correct signature and reading the fuse bytes also work.
Interestingly enough with the lock bit being 0xFF writing to flash using SPM still does not work.

alex-dewar, 2021-06-21:
Alright, I think I've fixed this with commit acceb1f56402412d38cf6a7490595f4ec7a09691

Turns out we weren't using the right ISP command to read the lock bits. I get 0xC0 when reading the lock bits, which I think is correct: there's two lock bits, so this would mean they're both 1.

alex-dewar, 2021-06-28:
I'm going to close this, as I believe this issue has been fixed.

closed 2021-06-28 by alex-dewar