newaetech/chipwhisperer #348: directory problem of libusb in conda pip install

mcsmonk, 2021-07-15:
With conda 4.10.1 & windows 10, I installed chipwhisperer==5.5.2 using pip
Then, usb backend DLL file is installed in the below path.
C:\Users\$USERNAME$\anaconda3\envs\cw\chipwhisperer\hardware\naeusb\libusb-1.0.dll

However, naeusb found backend in the belows,
``` python
if backend == "libusb1":
   import pathlib
   libusb_dir = pathlib.Path(__file__).parent.absolute()
   # C:\Users\$USERNAME$\anaconda3\envs\cw\lib\site-packages\chipwhisperer\hardware\naeusb
   libusb_path = pathlib.Path(libusb_dir, "libusb-1.0.dll")
   libusb_backend = libusb1.get_backend(find_library=lambda x: str(libusb_path))
if not libusb_backend:
   libusb_backend = libusb0.get_backend(find_library=lambda x: r"c:\Windows\System32\libusb0.dll")
```

And, it can't found backend.

I changed the location of the DLL file for now, but it seems good to fix this problem.

alex-dewar, 2021-07-15:
Thanks for the report on this. When you say you changed the location of the DLL, where did you move it to?

Anyway, I'm making this as "Won't Fix" as we're changing from pyusb to libusb1 in the next release, which comes bundled with libusb-1.0.dll.

mcsmonk, 2021-07-16:
I changed manually by choosing one out of two possible directories.
i.e., c:\Windows\System32\libusb0.dll or C:\Users\$USERNAME$\anaconda3\envs\cw\lib\site-packages\chipwhisperer\hardware\naeusb

I'm not understood how pip deal setup.py. But, a simple solution is to add code that cover the path in top directory of the environment.
i.e., Trying to find some directories

And, it might seem good to add a warning message on no finding backend dll.

alex-dewar, 2021-08-20:
I've tried replicating this (installing chipwhisperer in anaconda) and couldn't get it to happen, but I have seen others have the same issue, so I'm not too sure why this is happening.

Anyway, closing as we no longer use pyusb or libusb-1.0.dll

closed 2021-08-20 by alex-dewar