newaetech/chipwhisperer #349: SimpleSerial2 / _unstuff_data : sentinel limit does not depend on packet size

Arnaud-de-Grandmaison-ARM, 2021-07-16:
Hi,

While using the SimpleSerial2, I bumped on an issue while unstuffing packets.

In my case, the packets are 32 bytes long, and can be 32 x (unstuffed) 0, which means that the sentinel in [__unstuff_data](https://github.com/newaetech/chipwhisperer/blob/3bb8cb880b8616595841fa12706eea759e739fc3/software/chipwhisperer/capture/targets/SimpleSerial2.py#L154) will fire unduly.

I believe it would be better not to hard-code an arbitrary limit value (such as 30), but to instead use the buffer lenght (l) which could give a reasonable and sound upper limit for sentinel yet be suitable to the current packet size.

Kind regards,
Arnaud

alex-dewar, 2021-08-04:
Hi,

Sorry for getting to this so late, this should be fixed on the Husky branch by https://github.com/newaetech/chipwhisperer/commit/d4a70b885306d595b203914cf9bae6a6fe2e7843. I'll close this issue when we merge that in/do our next release.

Alex

alex-dewar, 2021-08-13:
Fixed by d4a70b885306d595b203914cf9bae6a6fe2e7843

closed 2021-08-13 by alex-dewar