newaetech/chipwhisperer #350: Empty Setup_Generic.ipynb file for Jupyter Notebook VM in BlackHat2021 Release

JeremyV2014, 2021-07-30:
I've encountered an issue trying to run the first step of _Section 1.2  ChipWhisperer Hardware Test_ in the _ChipWhisperer Setup Test_ jupyter notebook. I'm using the release version of the Jupyter Notebook VM for BlackHat 2021. I get this [output](https://github.com/newaetech/chipwhisperer/files/6909138/SetupCommandOutput.txt) when running the command.

I get the following error when I try to view _Setup_Generic.ipynb_ in Chrome:

[image]

Opening the file on the VM in nano shows that it is completely empty:

[image]

I tried re-downloading and extracting the VM and ran into the same issue. Can someone correct this?

colinoflynn, 2021-07-30:
Hmm - sorry we ran into this & thought it was limited to a few people, it seems the file got zerod at some point.

If you run a cell with this it should work:

```
!git checkout Setup_Scripts/Setup_Generic.ipynb
```

Basically need to re-check-out that file and it seems to fix it.

JeremyV2014, 2021-07-30:
@colinoflynn That fixed the issue for me. Thanks!

alex-dewar, 2021-07-30:
Going to reopen this until the end of Black Hat so that people can find it a bit easier.

closed 2021-08-04 by alex-dewar