newaetech/chipwhisperer #361: Install with `git submodule update --init --recursive` fails

vogelpi, 2021-08-17:
I just tried to do a clean install of chipwhisperer and noted that when doing a `git submodule update --init --recursive` on the current `develop` branch, git aborts with the following message
```
fatal: No url found for submodule path 'hardware/victims/cw305_artixtarget/fw/sam3u/CW305_SAM3U_FW/src/naeusb' in .gitmodules
```
I don't know if `naeusb` is really a submodule and the corresponding entry in `.gitmodules` is just missing or what the exact problem is. If I do a `git rm --cached hardware/victims/cw305_artixtarget/fw/sam3u/CW305_SAM3U_FW/src/naeusb` I can actually run the `git submodule --init` and install chipwhisperer, but this doesn't feel right to me.

colinoflynn, 2021-08-17:
Problem won't affect you for now - looking into it. The submodule in question is related to a major update to the USB firmware source code - we submodule'd some shared code (which I had Ctrl-C/Ctrl-V'd before, and Alex has been removed this duplication now with a single submodule). So your hacky fix won't affect the Python usage...

vogelpi, 2021-08-17:
Great, thanks @colinoflynn for the quick response!

alex-dewar, 2021-08-17:
Not sure why that module didn't get added to `.gitmodules`, but it should be fixed now.

vogelpi, 2021-08-18:
Thanks @alex-dewar for the quick fix. It works and I am thus closing this issue.

closed 2021-08-18 by vogelpi