newaetech/chipwhisperer #362: `target.simpleserial_write` seems broken on the CW-Lite

vogelpi, 2021-08-17:
Hi,
after reinstalling chipwhisperer from scratch today, the `target.simpleserial_write()` commands don't seem to work on the CW-Lite anymore. Doing e.g.
```
target.simpleserial_write("k", key)
```
I get the following error:
```
  File "src/chipwhisperer/software/chipwhisperer/capture/targets/SimpleSerial.py", line 335, in simpleserial_write
    self.write(cmd)
  File "src/chipwhisperer/software/chipwhisperer/capture/targets/SimpleSerial.py", line 238, in write
    raise e
  File "src/chipwhisperer/software/chipwhisperer/capture/targets/SimpleSerial.py", line 235, in write
    self.ser.write(data)
  File "src/chipwhisperer/software/chipwhisperer/capture/targets/simpleserial_readers/_base.py", line 94, in write
    self.hardware_write(string)
  File "src/chipwhisperer/software/chipwhisperer/capture/targets/simpleserial_readers/cwlite.py", line 78, in hardware_write
    self.cwlite_usart.write(string)
  File "src/chipwhisperer/software/chipwhisperer/hardware/naeusb/serial.py", line 139, in write
    datatosend = min(datatosend, 128-self.in_waiting_tx())
  File "src/chipwhisperer/software/chipwhisperer/common/utils/util.py", line 428, in func_wrapper
    return func(self, *args, **kwargs)
  File "src/chipwhisperer/software/chipwhisperer/hardware/naeusb/serial.py", line 166, in in_waiting_tx
    data = self._usartRxCmd(self.USART_CMD_NUMWAIT_TX, dlen=4)
  File "src/chipwhisperer/software/chipwhisperer/hardware/naeusb/serial.py", line 211, in _usartRxCmd
    return self._usb.readCtrl(self.CMD_USART0_CONFIG, cmd | (self._usart_num << 8), dlen)
  File "src/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 540, in readCtrl
    return self.usbseralizer.readCtrl(cmd, value, dlen)
  File "src/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 317, in readCtrl
    response = self.handle.controlRead(0xC1, cmd, value, 0, dlen, timeout=self._timeout)
AttributeError: 'NoneType' object has no attribute 'controlRead'
```
Doing e.g. a
```
target.write('v' + '\n')
```
i.e., without using the SimpleSerial part, still works.

Maybe this is just a matter of how we are initializing the scope and target. So far, this has worked without issues:
```
scope = cw.scope()
target = cw.target(scope)
```

alex-dewar, 2021-08-17:
That's really strange! The only difference between `simpleserial_write()` and `write()` from a USB perspective is that `simpleserial_write()` is that the latter flushes the serial buffer. Maybe the USB object is somehow being destroyed
during the flush?

vogelpi, 2021-08-18:
Thanks @alex-dewar for your input! It seems indeed that the USB object isn't there anymore. It's probably related to segmented mode of the CW-Lite or how we have been using that on our end. After doing some more digging, I noted that we actually do the following:

1. Init scope and target using:
```
scope = cw.scope()
target = cw.target(scope)
```
2. Disconnect the scope using `scope.dis()`
3. Re-initialize the scope in segmented mode

If I try to do another `target = cw.target(scope)` after 3. I get
```
  File "chipwhisperer/software/chipwhisperer/capture/targets/simpleserial_readers/cwlite.py", line 62, in con
    ser = scope._cwusb
AttributeError: 'CwLiteSegmented' object has no attribute '_cwusb'
```
So it seems we now miss some steps during the [re-initialization part in segmented mode](https://github.com/lowRISC/ot-sca/blob/master/cw/cw305/cw_lite_segmented.py). I know this is not your code, but if you have any clues, this would be appreciated.

vogelpi, 2021-08-18:
FYI @alphan

alex-dewar, 2021-08-18:
Try doing:

```python
cw_lite = CwLiteSegmented()
target = cw.target(cwlite._scope)
```

When you do `scope.dis()`, it destroys the USB connection that the target needs to send things over UART, so you need to provide it with the new scope object so it can get the new USB handle.