newaetech/chipwhisperer #367: CWLITEXMEGA: Programming target board results in LibUSB_ERROR_PIPE

Securitybits-io, 2021-09-15:
Trying to start the courses (Though this issue might be better to the chipwhisperer-jupyter repo), right now Fault 101 1_1, but get to the third code block where you upload the firmware.hex to the target board but the resulting message is ```USBErrorPipe: LIBUSB_ERROR_PIPE [-9]```

OS: Kali Linux Rolling
Board: CWLITEXMEGA
CW FW: 
```
Python 3.9.7 (default, Sep  3 2021, 06:18:44) 
[GCC 10.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import chipwhisperer as cw
>>> cw.scope()
cwlite Device
sn         = 5[..Nope..]5
fw_version = 
    major = 0
    minor = 50
    debug = 0

[... snip ...]
```
Dependencies:
```
libusb-1.0-0-dev is already the newest version (2:1.0.24-3).
libusb-dev is already the newest version (2:0.1.12-32).
make is already the newest version (4.3-4.1).
```

Now im kind of at a loss and i am dumping the full error down below, so beware of the wall of text in order to trace the error correctly.
```
---------------------------------------------------------------------------
USBErrorPipe                              Traceback (most recent call last)
<ipython-input-10-20c0cacb1c9e> in <module>
      1 fw_path = "../../../hardware/victims/firmware/simpleserial-glitch/simpleserial-glitch-{}.hex".format(PLATFORM)
----> 2 cw.program_target(scope, prog, fw_path)

~/Documents/tools/chipwhisperer/software/chipwhisperer/__init__.py in program_target(scope, prog_type, fw_path, **kwargs)
     75         prog._logging = None
     76         prog.open()
---> 77         prog.find()
     78         prog.erase()
     79         prog.program(fw_path, memtype="flash", verify=True)

~/Documents/tools/chipwhisperer/software/chipwhisperer/capture/api/programmers.py in func_wrapper(self, *args, **kwargs)
     69             self.scope.io.nrst = 'high_z'
     70         try:
---> 71             val = func(self, *args, **kwargs)
     72         finally:
     73             target_logger.debug('Restoring pdic, pdid, and nrst pin configuration')

~/Documents/tools/chipwhisperer/software/chipwhisperer/capture/api/programmers.py in find(self, xmega)
    224     def find(self, xmega=None):
    225         xmega = self.xmegaprog()
--> 226         sig, chip = xmega.find()
    227 
    228         # Print signature of unknown device

~/Documents/tools/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py in find(self)
    182 #### HIGH LEVEL FUNCTIONS
    183     def find(self):
--> 184         self.setParamTimeout(100)
    185         self.enablePDI(True)
    186 

~/Documents/tools/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py in setParamTimeout(self, timeoutMS)
    304         pload = [self.XPROG_PARAM_TIMEOUT]
    305         pload.extend(packuint32(timeoutticks))
--> 306         self._xmegaDoWrite(self.XPROG_CMD_SET_PARAM, pload)
    307 
    308 

~/Documents/tools/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py in _xmegaDoWrite(self, cmd, data, checkStatus)
    512 
    513         # windex selects interface
--> 514         self._usb.sendCtrl(self.CMD_XMEGA_PROGRAM, cmd, data)
    515 
    516         # Check status

~/Documents/tools/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py in sendCtrl(self, cmd, value, data)
    547         """
    548         # Vendor-specific, OUT, interface control transfer
--> 549         self.usbseralizer.sendCtrl(cmd, value, data)
    550 
    551     def readCtrl(self, cmd, value=0, dlen=0):

~/Documents/tools/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py in sendCtrl(self, cmd, value, data)
    323                     bRequest: {:02X}, wValue: {:04X}, wIndex: {:04X}, data: {}".format(0x41, cmd, \
    324                         value, 0, data))
--> 325         self.handle.controlWrite(0x41, cmd, value, 0, data, timeout=self._timeout)
    326         #return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
    327 

/usr/lib/python3/dist-packages/usb1/__init__.py in controlWrite(self, request_type, request, value, index, data, timeout)
   1347         # pylint: enable=undefined-variable
   1348         data, _ = create_initialised_buffer(data)
-> 1349         return self._controlTransfer(request_type, request, value, index, data,
   1350                                      sizeof(data), timeout)
   1351 

/usr/lib/python3/dist-packages/usb1/__init__.py in _controlTransfer(self, request_type, request, value, index, data, length, timeout)
   1324             timeout,
   1325         )
-> 1326         mayRaiseUSBError(result)
   1327         return result
   1328 

/usr/lib/python3/dist-packages/usb1/__init__.py in mayRaiseUSBError(value, __raiseUSBError)
    125     ):
    126     if value < 0:
--> 127         __raiseUSBError(value)
    128     return value
    129 

/usr/lib/python3/dist-packages/usb1/__init__.py in raiseUSBError(value, __STATUS_TO_EXCEPTION_DICT, __USBError)
    117         __USBError=USBError,
    118     ):
--> 119     raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
    120 
    121 def mayRaiseUSBError(

USBErrorPipe: LIBUSB_ERROR_PIPE [-9]
```

alex-dewar, 2021-09-15:
Hi,

It looks like firmware 0.50 has disabled XMEGA/AVR programming. If you revert to an earlier firmware, it should fix the issue.

We'll get this solved for the next release.

keesj, 2021-09-16:
@Securitybits-io Thanks. for reporting I was about to create an issue myself!

Securitybits-io, 2021-09-16:
@keesj Awesome, can confirm that the solution works for now!

alex-dewar, 2021-09-24:
The latest firmware on develop also now has this issue fixed

closed 2022-02-11 by alex-dewar