newaetech/chipwhisperer #379: CW305: `fpga_write`/`fpga_read` not working for payloads longer than 1024 bytes

jonnykl, 2021-12-14:
`fpga_write()` and `fpga_read()` are not working correctly when using payloads longer than 1024 bytes. This issue is similar to my previous issue #373. Again I map the address space to a memory on the FPGA. I could see that something is written/read to/from the memory. The written/read data is partially correct but the most bytes are wrong. As a temporary fix, I split the written/read data in chunks of 1024 bytes and call `fpga_write`/`fpga_read` multiple times.

jpcrypt, 2021-12-14:
What is your `pBYTECNT_SIZE` parameter? If it's 10, then this is the expected behaviour. See e.g.:
https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/cw305_artixtarget/fpga/common/cw305_usb_reg_fe.v#L81

jonnykl, 2021-12-14:
I'm using my own FPGA implementation which handles the communication. As in my previous issue #373, I'm setting `target.bytecount_size` to `0`. So `target.fpga_write(0, [0xAA]*2048)` should do 2048 writes to the addresses from 0 to 2047 with the value `0xAA`. This works for payloads which are not longer than 1024 bytes. So this must be a bug in the SAM3U FW or the Python code run on the host.

jpcrypt, 2021-12-14:
Sorry, I had missed that detail.
But, setting `target.bytecount_size` won't change the Verilog `pBYTECOUNT_SIZE`; what is that set to? (or have you changed the Verilog logic around that too?).

alex-dewar, 2021-12-14:
Hi, thanks for the report on this. Yeah it looks like there was a bug where writes > 1024 would repeatedly write to the first 1024 bytes. That should fixed now. I didn't see any issue with the read code, so the issue with the read might just be as a consequence of the invalid write.

This should be fixed in the latest commit, but I don't have a bitstream that accepts reads that long. Do you mind testing and seeing if it fixes the issue?

jonnykl, 2021-12-14:
> Sorry, I had missed that detail. But, setting `target.bytecount_size` won't change the Verilog `pBYTECOUNT_SIZE`; what is that set to? (or have you changed the Verilog logic around that too?).

I'm not using the verilog code from this repo, so I have no `pBYTECOUNT_SIZE` parameter.

> This should be fixed in the latest commit, but I don't have a bitstream that accepts reads that long. Do you mind testing and seeing if it fixes the issue?

Thanks, it works now (tested both `fpga_write` and `fpga_read` with a payload of 8192 bytes).

closed 2021-12-14 by jonnykl

alex-dewar, 2021-12-14:
Oh yeah, we just embed the binary in https://github.com/newaetech/chipwhisperer/blob/develop/software/chipwhisperer/hardware/firmware/cw305.py. The actual fix was at https://github.com/newaetech/naeusb/tree/main, but I forgot to update the submodule, so you didn't see it in Git.