newaetech/chipwhisperer #380: CW-Husky: First trace in segmented mode off by one sample

vogelpi, 2022-01-03:
Hi @colinoflynn, @jpcrypt ,
I just noted that when performing a segmented capture, it seems the the first trace of every segment seems to be shifted in time (off by one sample). For example, this is what I get when plotting the first 10 traces:
[image]

When using the normal/non-segmented capture, everything seems to be fine.

FYI, I am using chipwhisperer 723c0c01763b9cc7cf7331d5dda8ca54ee82bc99 and also upgraded Husky to the latest version.

jpcrypt, 2022-01-03:
Thanks for the report, I will look into this. Are you using `scope.segment_cycle_counter_en`?

vogelpi, 2022-01-03:
I think we don't change that variable. From the description I assume it's 0. What we do is arming the scope once and then we get one trigger event per segment. Thanks for looking into it @jpcrypt !

vogelpi, 2022-01-04:
By the way, the problem doesn't exist with chipwhisperer 5997c2400544c38c5cce39ef9c2ac277433c8a5c. It seems to have been introduced sometime between end of Nov and mid Dec.

vogelpi, 2022-01-04:
> Hm this is really strange. I now switched to another machine and CW305 instead of CW310 and there the problem doesn't seem to exist.

Correction. The problem is still there but it doesn't seem to be just the first trace on the other machine/CW305.

jpcrypt, 2022-01-04:
Hmm strange. I'm seeing the first trace problem (just like in your plot above) when `scope.adc.offset` is not zero; I have a potential fix for it but I'm still doing some testing for it. If you can share your `scope.adc` settings I'll make sure there isn't anything else I've missed.

vogelpi, 2022-01-04:
We use the following two routines to setup the scope:
1. https://github.com/lowRISC/ot-sca/blob/5997c2400544c38c5cce39ef9c2ac277433c8a5c/cw/cw305/util/device.py#L110-L136
2. https://github.com/lowRISC/ot-sca/blob/5997c2400544c38c5cce39ef9c2ac277433c8a5c/cw/cw305/cw_segmented.py#L169-L188

`scope.adc.offset` should be zero in our case (second snippet).

closed 2022-01-05 by jpcrypt

vogelpi, 2022-01-06:
Thanks a lot @jpcrypt , this indeed solves the problem!