newaetech/chipwhisperer #381: Can't import libusb on MacBook M1

MrUndead13, 2022-01-04:
Good afternoon,
I'm trying to use a chipwhisperer lite on my MacBook Air m1 and I can't connect both.
When I run the Connecting to Hardware file, I get an error after running:

import chipwhisperer as cw
scope = cw.scope()

The error is:

OSError: dlopen(/opt/homebrew/opt/libusb/lib/libusb-1.0.dylib, 0x0006): tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.dylib' (no such file), '/usr/lib/libusb-1.0.dylib' (no such file), '/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file)

and at the end:

OSError: Could not import libusb dll. Try 
pip uninstall libusb1
pip install libusb1

even if I installed libusb and put a link in usr/local/lib/libusb to opt/homebrew/opt/libusb

Is it possible to use it on ARM and is there a fix to that or not at all ?

colinoflynn, 2022-01-04:
We don't have a machine to test on right now easily but can bring a M1 up. Unfortunately due to current pandemic we're not in the normal office so things are much slower...

If you see the tests described in https://github.com/pyusb/pyusb/issues/355 do they seem to work?

Is there a full stack trace (showing where it's getting called from)? That `incompatible architecture` error makes it seem like it's not picking up the correct version, which also looks to be a problem mentioned in that issue.

As a quick test if you open the Python used for ChipWhisperer and run:

```
import usb.core
dev = usb.core.find()
```
And see if that gives you the "no backend" problem or seems to work? I *think* the above should work - if not see the thread on that pyusb link, which has some different things to try.

MrUndead13, 2022-01-04:
No problem at all ! You still take time for my issue and I thank you a lot for that  !(while others keep redirecting me to other repositories and close without asking anything....)

After running:
$ python --version
$ python -c "import ctypes.util; print(ctypes.util.find_library('usb'))"

I get:
Python 3.9.7
/usr/local/lib/libusb.dylib

I wasn't able to understand how to create a find_library method but I created the link between /usr/local and my homebrew folder and it still doesn't work unfortunately.

I will write the full trace in the next message :)

import usb.core
dev = usb.core.find()
gives me no backend error so I'l look into the example !

MrUndead13, 2022-01-04:
Full trace:

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
~/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/jupyter/Setup_Scripts/Setup_Generic.ipynb in <module>
      3 try:
----> 4     if not scope.connectStatus:
      5         scope.con()

NameError: name 'scope' is not defined

During handling of the above exception, another exception occurred:

OSError                                   Traceback (most recent call last)
~/opt/anaconda3/lib/python3.9/site-packages/usb1/_libusb1.py in __getLibrary()
    157     try:
--> 158         return dll_loader(filename, use_errno=True, use_last_error=True)
    159     except OSError:

~/opt/anaconda3/lib/python3.9/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    381         if handle is None:
--> 382             self._handle = _dlopen(self._name, mode)
    383         else:

OSError: dlopen(libusb-1.0.dylib, 0x0006): tried: '/Users/nathanpaulet/opt/anaconda3/lib/libusb-1.0.dylib' (no such file), '/Users/nathanpaulet/opt/anaconda3/lib/libusb-1.0.dylib' (no such file), '/Users/nathanpaulet/opt/anaconda3/lib/python3.9/lib-dynload/../../libusb-1.0.dylib' (no such file), '/Users/nathanpaulet/opt/anaconda3/lib/libusb-1.0.dylib' (no such file), '/Users/nathanpaulet/opt/anaconda3/bin/../lib/libusb-1.0.dylib' (no such file), 'libusb-1.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.dylib' (no such file), '/usr/lib/libusb-1.0.dylib' (no such file), '/Users/nathanpaulet/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/jupyter/libusb-1.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.dylib' (no such file), '/usr/lib/libusb-1.0.dylib' (no such file)

During handling of the above exception, another exception occurred:

OSError                                   Traceback (most recent call last)
~/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py in __init__(self)
    289             self.usb_ctx = usb1.USBContext()
--> 290             self.usb_ctx.open()
    291         except OSError as e:

~/opt/anaconda3/lib/python3.9/site-packages/usb1/__init__.py in open(self)
   2174         assert not self.__context_p
-> 2175         loadLibrary()
   2176         self.__libusb_handle_events = libusb1.libusb_handle_events

~/opt/anaconda3/lib/python3.9/site-packages/usb1/_libusb1.py in loadLibrary(libusb)
    209             if not __loaded: # we were not loaded, check again under lock
--> 210                 __loadLibrary(libusb)
    211                 __loaded = True

~/opt/anaconda3/lib/python3.9/site-packages/usb1/_libusb1.py in __loadLibrary(libusb)
    218     if libusb is None:
--> 219         libusb = __getLibrary()
    220 

~/opt/anaconda3/lib/python3.9/site-packages/usb1/_libusb1.py in __getLibrary()
    183                 raise
--> 184         return dll_loader(libusb_path, use_errno=True, use_last_error=True)
    185 

~/opt/anaconda3/lib/python3.9/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    381         if handle is None:
--> 382             self._handle = _dlopen(self._name, mode)
    383         else:

OSError: dlopen(/opt/homebrew/opt/libusb/lib/libusb-1.0.dylib, 0x0006): tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.dylib' (no such file), '/usr/lib/libusb-1.0.dylib' (no such file), '/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file)

The above exception was the direct cause of the following exception:

OSError                                   Traceback (most recent call last)
~/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/jupyter/Setup_Scripts/Setup_Generic.ipynb in <module>
      5         scope.con()
      6 except NameError:
----> 7     scope = cw.scope()
      8 
      9 try:
[... 142 lines cut ...]

~/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/software/chipwhisperer/__init__.py in scope(scope_type, name, sn, idProduct, bitstream, force, prog_speed, **kwargs)
    325 
    326     if scope_type is None:
--> 327         scope_type = get_cw_type(**kwargs)
    328     rtn : scopes.ScopeTypes = scope_type()
    329     try:

~/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/software/chipwhisperer/common/utils/util.py in get_cw_type(sn, idProduct, hw_location, **kwargs)
    517         possible_ids = [0xace0, 0xace2, 0xace3, 0xace5]
    518 
--> 519     cwusb = NAEUSB_Backend()
    520     device = cwusb.find(serial_number=sn, idProduct=possible_ids, hw_location=hw_location)
    521     name = device.getProduct()

~/Desktop/Travail/DD/Cours/Master Thesis/Side Channels/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py in __init__(self)
    292             # naeusb_logger.error("Could not import libusb dll. Try pip uninstall libusb1, then pip install libusb1")
    293             self.usb_ctx = None
--> 294             raise OSError("Could not import libusb dll. Try \npip uninstall libusb1\npip install libusb1") from e
    295         self.handle = None
    296         self.device = None

OSError: Could not import libusb dll. Try 
pip uninstall libusb1
pip install libusb1

alex-dewar, 2022-01-04:
That error message says you need an x86_64 lib. I'm not very familiar with the M1, but I believe it has an x86 emulation mode, right? It might be that your Python version is x86, but you have an Arm version of libusb1.

alex-dewar, 2022-01-04:
I think you can check via Activity Monitor. If the python process is of type "Apple", then it's running natively. It should show up as "Intel" if it's x86.

MrUndead13, 2022-01-04:
It says Intel !!!!
SO I need to find a way to switch to the Apple version

alex-dewar, 2022-01-04:
It looks like https://github.com/conda-forge/miniforge has a native build for the M1.

MrUndead13, 2022-01-05:
Thank you I'll look into that !

alex-dewar, 2022-02-04:
Closing, as I believe this issue has been resolved

closed 2022-02-04 by alex-dewar

TnnsBeast, 2023-08-07:
Could you provide some information as to how you fixed this problem? I am experiencing the same issue on M1 Mac even after using native M1 miniforge

alex-dewar, 2023-08-07:
I think one thing that came up is that you need to make sure you're using the native version of libusb as well. I'd recommend grabbing the latest homebrew and using that to get libusb.