newaetech/chipwhisperer #402: Install through pip in regular/non-editable mode seems broken

vogelpi, 2022-05-12:
Hi,
for a while we have been installing and updating ChipWhisperer in non-editable mode through pip, i.e., with `pip install --user chipwhisperer`. Recently (don't exactly know when) this stopped working. We are now seeing the following error:
```console
ERROR:ChipWhisperer Scope:ChipWhisperer error state detected. Resetting and retrying connection...
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.8/site-packages/chipwhisperer/__init__.py", line 337, in scope
    rtn.con(**kwargs)
  File "/home/user/.local/lib/python3.8/site-packages/chipwhisperer/capture/scopes/OpenADC.py", line 550, in con
    self.trace = TraceWhisperer(husky=True, target=None, scope=self, trace_reg_select=3, main_reg_select=2)
  File "/home/user/.local/lib/python3.8/site-packages/chipwhisperer/capture/trace/TraceWhisperer.py", line 130, in __init__
    self.slurp_defines(defines_files, trace_reg_select, main_reg_select)
  File "/home/user/.local/lib/python3.8/site-packages/chipwhisperer/capture/trace/TraceWhisperer.py", line 216, in slurp_defines
    defines = open(defines_file, 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/.local/lib/python3.8/site-packages/chipwhisperer/capture/trace/defines/defines_trace.v'
```
If we install ChipWhisperer in editable mode things are still working.

alex-dewar, 2022-05-12:
It might be because those files aren't in https://github.com/newaetech/chipwhisperer/blob/develop/setup.py. I'll add them and see if that fixes it

alex-dewar, 2022-05-19:
Alright, I think I've got this fixed in the current setup.py. I've confirmed that installing as a user wouldn't move those files, added them, then confirmed they were present now. 

I'll close this when we push out the new release.

vogelpi, 2022-05-19:
Thanks @alex-dewar for taking care of this!

closed 2022-06-07 by alex-dewar