newaetech/chipwhisperer #405: Warnings when ADC operated close to max sampling rate

vogelpi, 2022-05-19:
Hi,
we've recently adjusted our capture setup to make best use of Husky. To this end, we've doubled the sampling rate from previously 100 MS/s to 200 MS/s. The scope is using the target clock (100 MHz) and multiplies that internally to obtain the 200 MHz sampling clock. Now we get quite some warnings like this:
```console
WARNING:ChipWhisperer Scope:
ADC frequency exceeds specification (200 MHz).
This may or may not work, depending on temperature, voltage, and luck.
It may not work reliably.
You can run scope.adc_test() to check whether ADC data is sampled properly by the FPGA,
but this doesn't fully verify that the ADC is working properly.
```
When reading out the effective sampling rate we get e.g.
```console
Scope setup with sampling rate 200009468.0 S/s
```
I think this sampling rate should still be fine. At least the traces look good and we can detect leakage and also perform attacks. So maybe it's just that the function checking the ADC frequency needs a little bit of extra margin to suppress the warning in this case.

colinoflynn, 2022-05-19:
You're 9468 S/s over, works as intended ;-) FYI @jpcrypt just did a nice update to the FPGA that also fully meets timing as well (I think up to 250 MHz, @jpcrypt let me know if I was remembering wrong).

We had been experimenting more and we think the ADC is also good up to faster than specified frequencies, they have a version with a 250 MS/s rating that we sub in depending on stock. I suspect the 200 MS/s and 250 MS/s is just binning etc, and thus you can use even the 200 MS/s rated ADC well above that.

But anyway to this bug - I think you are right makes sense to set this warning to kick on at 200.9 MS/s or something to avoid the warning.

closed 2022-05-19 by jpcrypt

colinoflynn, 2022-05-19:
> would fail if the FPGA were operating at the extreme limits of its allowed range

I mean the "Husky" part of ChipWhisperer-Husky implies it's designed to work well in cold conditions, so anybody who uses this in the Australian outback should get exactly what they expect!

vogelpi, 2022-05-20:
Thanks guys for taking care of this so quickly!