newaetech/chipwhisperer #413: simpleserial-rsa hard fault on STM32

jpcrypt, 2022-07-05:
[This commit](https://github.com/newaetech/chipwhisperer/commit/c73f434f509c3b7b4ee4d357361dc7c5ffbd84b4) seems to be the culprit. When this is rolled back, things work fine with `SS_VER_2_1`, but not `SS_VER_1_1`.

alex-dewar, 2022-07-05:
I've reverted this commit for now. Not too sure what's causing the crash yet, but it doesn't seem like rand crash happens commonly - the function can run many times without crashing, but is called many times in the RSA calculation.

alex-dewar, 2022-07-05:
It seems like the location for the stack was setup incorrectly and was being overwritten when the rand initialization function was called, eventually causing the device to pop a garbage value into the PC. It should now be fixed.

twelho, 2023-12-12:
For reference, https://github.com/newaetech/chipwhisperer/commit/392807cc84db92f073e62c42b1f018d09cff2249 reverts https://github.com/newaetech/chipwhisperer/commit/c73f434f509c3b7b4ee4d357361dc7c5ffbd84b4 and then https://github.com/newaetech/chipwhisperer/commit/13f37a6406e7ef7f8a92ff1f62f0b4a7bc85ba74 fixes the stack location. Just hit this on 5.6.1, will try again with 5.7.0 that has these fixes.