newaetech/chipwhisperer #414: Glitch output with very low voltage level

itewqq, 2022-07-15:
Hi there,

I was trying to use Chipwhisperer-Lite and ChipSHOUTER to work together, but found that Chipwhisperer could not drive the latter's hardware trigger.

Then I connected the glitch output of Chipwhisperer to the oscilloscope and found that **it can only output a very low voltage (0.4V)** at a time, is this normal? How can I get it to work properly?

[image]

For the hardware setup, I just plug the glitch output to a picoscope:

[image]

colinoflynn, 2022-07-15:
Hi @itewqq - the gltich output is actually one of two MOSFETs on the output you are activating, so there is no high drive. If you want to generate a logic-level signal, you'll have to route it to HS2:

```
scope.glitch.hs2 = "glitch"
```

However - you often want to use HS2 for clock & not glitch. This can be done using a pull-up on the glitch output & enabling the `glitch_lp` MOSFET.

But if using ChipSHOUTER, you don't need any external hardware as you can configured it to be active-low on trigger input, and it by default has a pull-up to prevent the input from floating. To do this, two configurations needed:

On ChipWhisperer:
```
scope.glitch.output = "enable_only"
scope.io.glitch_lp = True
```

On ChipSHOUTER:
```
set hwtrig_term 0
set hwtrig_mode 0
```
Note you might want to save those as new defaults with `write_nvm`, otherwise they need to be reconfigured on ever reboot of ChipSHOUTER.

[image]

This is described in more detail on Page 26 of the [ChipSHOUTER User Manual](https://media.newae.com/manuals/ChipSHOUTER_PRESS_1.4.pdf) (PDF warning).

Hope this helps! It's a little non-obvious about what the glitch output is, easy documentation is always lagging. Let us know if that works ok + makes sense.

itewqq, 2022-07-16:
Thanks for the explanation! 

> This is described in more detail on Page 26 of the [ChipSHOUTER User Manual](https://media.newae.com/manuals/ChipSHOUTER_PRESS_1.4.pdf) (PDF warning).

Actually I've tried to connect ChipWhiperer and ChipSHOUTER the same way as described in the User Manual, but when I use `scope.glitch.manual_trigger()` in ChipWhisperer nothing happends. Besides the ChipSHOUTER may crash after this.

> On ChipWhisperer:
> 
> scope.glitch.output = "enable_only"
> scope.io.glitch_lp = True
> On ChipSHOUTER:
> 
> set hwtrig_term 0
> set hwtrig_mode 0

and the configuration I've adopted is exactly same here.

itewqq, 2022-07-18:
Also, when I try to use `scope.glitch.hs2`, there is an error:

```
In [3]: scope.glitch.hs2 = "glitch"
(ChipWhisperer Other ERROR|File util.py:362) Setting unknown attribute hs2 in <class 'chipwhisperer.capture.scopes.cwhardware.ChipWhispererGlitch.GlitchSettings'>

```

itewqq, 2022-07-18:
Ok now I'm not able to connect chipwhisperer :sweat_smile:

itewqq, 2022-07-18:
The current version my test code is:

```python
from chipshouter import ChipSHOUTER
import chipwhisperer as cw
import time

CS_COM = "/dev/ttyUSB0"
cs = ChipSHOUTER(CS_COM)
cs.hwtrig_term = 0
cs.hwtrig_mode = 0

scope = cw.scope()
scope

scope.io.glitch_lp = True
scope.io.glitch_hp = False
scope.glitch.output = 'enable_only'
scope.glitch.repeat = 3
scope.glitch.trigger_src = 'manual'

print(scope.glitch)
cs.armed = True
time.sleep(1)

scope.arm()
scope.glitch.manual_trigger()
```

BTW I can trigger the ChipSHOUTER with a raspberry pi 3b+ using gpio pin.

The hardware connection is just same as you mentioned:

[image]

alex-dewar, 2022-07-18:
I'll try replicating this on a ChipShouter. Have you tried measuring the trigger pin on the ChipShouter to see if the pullup is actually pulling the IO line up? I'm fairly sure that's the issue.

For clock glitching, it should be `scope.io.hs2 = "glitch"` instead of `scope.glitch.hs2 = "glitch"`. Also you should set `scope.glitch.clk_src = "clkgen"` to setup the glitch clock.

Also sorry for the version check, I've modified it so it should only check once, then timeout (default is 5 times).

colinoflynn, 2022-07-18:
Oops sorry forgot to respond as well - I'd double-check the glitch clock is running too. You can issue a reset to the glitch clock DCM, it may be the glitch clock isn't running. I'd crank the `repeat` way up as a first test too (set it to 500 or something).

itewqq, 2022-07-20:
>Have you tried measuring the trigger pin on the ChipShouter to see if the pullup is actually pulling the IO line up? I'm fairly sure that's the issue.

Yes, the trigger pin of ChipShouter is actually up.

>For clock glitching, it should be scope.io.hs2 = "glitch" instead of scope.glitch.hs2 = "glitch". Also you should set scope.glitch.clk_src = "clkgen" to setup the glitch clock.

I've set the scope.glitch.clk_src = "clkgen" but it didn't work, too. My test code:

```python
from chipshouter import ChipSHOUTER
import chipwhisperer as cw

CS_COM = "/dev/ttyUSB1"

cs = ChipSHOUTER(CS_COM)
scope = cw.scope()

cs.hwtrig_term = 0
cs.hwtrig_mode = 0

scope.io.glitch_lp = True
scope.io.glitch_hp = False
scope.glitch.clk_src = "clkgen" # set glitch input clock
scope.glitch.output = 'enable_only'
scope.glitch.repeat = 500
scope.glitch.trigger_src = 'manual'

cs.armed = True
scope.arm()
scope.glitch.manual_trigger()
```

Also, when I first connect the ChipWhisperer and ChipSHOUTER, then run the code, ChipSHOUTER will alert with a beep and the lights are:

[image]

And the fault light won't go off until I disconnect the CW and CS.

itewqq, 2022-07-20:
I also tried the `hs2` output pin, it's not working too.

My test code:
```python
import chipwhisperer as cw
import time

scope = cw.scope()
scope

scope.glitch.output = 'enable_only'
scope.glitch.clk_src = "clkgen"
scope.io.hs2 = "glitch"
scope.glitch.repeat = 500
scope.glitch.trigger_src = 'manual'

scope.arm()
scope.glitch.manual_trigger()
```
Output of `scope`:

```
cwlite Device
sn         = 53313120313436373130303130333032
fw_version = 
    major = 0
    minor = 62
    debug = 0
gain = 
    mode = low
    gain = 0
    db   = 5.5
adc = 
    state          = False
    basic_mode     = low
    timeout        = 2
    offset         = 0
    presamples     = 0
    samples        = 24400
    decimate       = 1
    trig_count     = 902539329
    fifo_fill_mode = normal
clock = 
    adc_src       = clkgen_x1
    adc_phase     = 0
    adc_freq      = 96000000
    adc_rate      = 96000000.0
    adc_locked    = True
    freq_ctr      = 0
    freq_ctr_src  = extclk
    clkgen_src    = system
    extclk_freq   = 10000000
    clkgen_mul    = 2
    clkgen_div    = 1
    clkgen_freq   = 192000000.0
    clkgen_locked = True
trigger = 
    triggers = tio4
    module   = basic
io = 
    tio1         = serial_tx
    tio2         = serial_rx
    tio3         = high_z
    tio4         = high_z
    pdid         = high_z
    pdic         = high_z
    nrst         = high_z
    glitch_hp    = False
    glitch_lp    = False
    extclk_src   = hs1
    hs2          = glitch
    target_pwr   = True
    tio_states   = (1, 0, 0, 0)
    cdc_settings = bytearray(b'\x01\x00\x00\x00')
glitch = 
    clk_src     = clkgen
    width       = 10.15625
    width_fine  = 0
    offset      = 10.15625
    offset_fine = 0
    trigger_src = manual
    arm_timing  = after_scope
    ext_offset  = 0
    repeat      = 500
    output      = enable_only
```

My hardware connection:

[image]

alex-dewar, 2022-07-20:
Can you try calling `scope.default_setup()` as well? That should reset all the clocks and make sure the DCMs are locked

itewqq, 2022-07-23:
>Can you try calling scope.default_setup() as well? That should reset all the clocks and make sure the DCMs are locked

It's working perfectly after `scope.default_setup() `, thanks very much!!!

itewqq, 2023-12-03:
Well it happened AGAIN, after I update the firmware:
[image]

The output of the Glitch signal is too small to trigger the ChipSHOUTER, just as the above disccued issue.
But this time the `scope.default_setup()` trick does not work...

Test code:
```python
import chipwhisperer as cw

# ChipWhisperer
scope = cw.scope()
scope.default_setup()

# ChipWhisperer trigger settings

scope.io.glitch_lp = True
scope.io.glitch_hp = False
scope.glitch.clk_src = "clkgen" # set glitch input clock
scope.glitch.output = 'enable_only'
scope.glitch.repeat = 100
scope.glitch.trigger_src = 'manual'

# Manully trigger it
scope.arm()
```

Env:
```
cwlite Device
sn         = 53313120313436373130303130333032
fw_version = 
    major = 0
    minor = 64
    debug = 0
gain = 
    mode = high
    gain = 30
    db   = 24.8359375
adc = 
    state          = False
    basic_mode     = rising_edge
    timeout        = 2
    offset         = 0
    presamples     = 0
    samples        = 5000
    decimate       = 1
    trig_count     = 0
    fifo_fill_mode = normal
clock = 
    adc_src       = clkgen_x4
    adc_phase     = 0
    adc_freq      = 29538471
    adc_rate      = 29538471.0
    adc_locked    = True
    freq_ctr      = 0
    freq_ctr_src  = extclk
    clkgen_src    = system
    extclk_freq   = 10000000
    clkgen_mul    = 2
    clkgen_div    = 26
    clkgen_freq   = 7384615.384615385
    clkgen_locked = True
trigger = 
    triggers = tio4
    module   = basic
io = 
    tio1         = serial_rx
    tio2         = serial_tx
    tio3         = high_z
    tio4         = high_z
    pdid         = high_z
    pdic         = high_z
    nrst         = high_z
    glitch_hp    = False
    glitch_lp    = True
    extclk_src   = hs1
    hs2          = clkgen
    target_pwr   = True
    tio_states   = (0, 1, 0, 0)
    cdc_settings = bytearray(b'\x00\x00\x00\x00')
glitch = 
    clk_src     = clkgen
    width       = 10.15625
    width_fine  = 0
    offset      = 10.15625
    offset_fine = 0
    trigger_src = manual
    arm_timing  = after_scope
    ext_offset  = 0
    repeat      = 100
    output      = enable_only
```

Python package `chipwhisperer==5.7.0`.

itewqq, 2023-12-03:
@alex-dewar Thanks for the quick response!

I'm using the glitch SMA output, following the ChipShouter manul:

[image]

And it cannot trigger the ChipShouter so I connect the glitch SMA output to the oscilloscope to check it out.

>You need to use the HS2 pin with scope.io.hs2 = "glitch"

Do you mean that I should use the HS2 pin instead of the SMA connector? Or just write this in the code but still use the SMA output?

itewqq, 2023-12-03:
@alex-dewar Of course I did that, but seems not working.

I'm off my lab and will check it serveral hours later. 

BTW I'm always confused about the operation order. Currentyly my setting process is like:

1. Connect CW and CS in power off state.
2. Power on the CS and set `cs.hwtrig_term = 0`, `cs.hwtrig_mode = 0`
3. Power on the CW by connecting to computer and make a `scope.default_setup()`. 
4. Then setup other CW options like glitch_lp/glitch_hp.
5. Start my dedicate work hereafter.

Sometimes the CS won't be happy with this and beeps for 1~2 seconds. I wonder what's the correct way to boot everything up. Would be very appreciate if you can give some advice.

jpcrypt, 2023-12-04:
While CW-lite can generate a 200 MHz clock, its FPGA may not function properly at such a high clock frequency, i.e. it may not trigger and/or generate the output glitch pulse correctly (or at all).

itewqq, 2023-12-05:
Thanks @alex-dewar @jpcrypt

closed 2023-12-05 by itewqq