newaetech/chipwhisperer #416: Firmware build failure for CW305

cmomin, 2022-07-25:
Hello,

I am currently willing to add some features to the firmware of the CW305 board. However, I encounter some failure in the building process. More particularly, I have the following error message (Ubuntu 20.04.4 LTS (Focal Fossa))
```
"arm-none-eabi-gcc"    -c -o ../src/ASF/sam/boards/cw305/init.o ../src/ASF/sam/boards/cw305/init.c
../src/ASF/sam/boards/cw305/init.c:44:10: fatal error: compiler.h: No such file or directory
   44 | #include "compiler.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: ../src/ASF/sam/boards/cw305/init.o] Error 1
```

when I do a `make` under the path `chipwhisperer/hardware/victims/cw305_artixtarget/fw/sam3u/CW305_SAM3U_FW/build`.
This is obtained for a fresh clone of the repo on the `develop` branch. 

I was able to build the firmware previously (commit 872c73ac26b64b4c5dd5c9f5b4a7ad8d4ddc02e9 still working, I can try to dig for more recent commits), so I guess that it comes from some refactoring of the repo.

alex-dewar, 2022-07-25:
Hi,

We do most of our firmware build stuff through Atmel Studio, so I'd recommend that over the makefile. I'm pretty sure
that makefile needs to be updated (probably from a script) from the Atmel Studio project, so it probably got out of sync at some point. We also have a new submodule at `hardware/victims/firmware/cw305_artixtarget/fw/sam3u/CW305_SAM3U_FW/src/naeusb`, so you'll need to initialize+update that if you haven't already

That being said, I have been working on a simple makefile build system on the naeusb branch of ChipWhisperer. You should be able to build via:

```sh
# cd to chipwhisperer folder
cd hardware/victims/firmware/cw305_artixtarget/fw/sam3u/CW305_SAM3U_FW/src
git submodule update --init naeusb
make -j
```

I can't confirm that this builds working firmware yet, but it at least compiles.

cmomin, 2022-07-26:
Hi, 

Thanks for the fast answer. I'll try and keep you in touch!

cmomin, 2022-07-27:
Hi, 
I tried what you suggested and I still have the following compilation errors (from fresh clone again):

```
...
Linking: ChipWhisperer-CW305-SAM3U1C.elf
arm-none-eabi-gcc -mcpu=cortex-m3 -I. -fdata-sections -ffunction-sections -mlong-calls -g3 -Wall -pipe -fno-strict-aliasing -Wall -Wstrict-prototypes -Wmissing-prototypes -Wchar-subscripts -Wcomment -Wformat=2 --param max-inline-insns-single=500 -DDEBUG -D__SAM3U2E__ -DARM_MATH_CM3=true -Dprintf=iprintf -DUDD_ENABLE -Dscanf=iscanf -DPLATFORMCW1190=1 -gdwarf-2 -DHAL_TYPE=HAL_sam3u1c -DPLATFORM=SAM3U1C -DF_CPU=7372800UL -Og -funsigned-char -funsigned-bitfields -fshort-enums -Wall -Wno-missing-prototypes -Wno-unused-variable -Wno-pointer-sign -Wno-unused-value -Wstrict-prototypes -Wa,-adhlns=objdir/main_cw305.o -I./naeusb -I.././hal -I.././hal/sam3u1c -I.././hal/sam3u1c/inc -I.././hal/sam3u1c/inc/sam -I.././hal/sam3u1c/inc/sam_twi -std=gnu99 -MMD -MP -MF .dep/ChipWhisperer-CW305-SAM3U1C.elf.d objdir/main_cw305.o objdir/naeusb/usb_xmem.o objdir/naeusb/fpga_program.o objdir/naeusb/V2Protocol.o objdir/naeusb/V2ProtocolParams.o objdir/cdce906.o objdir/fpgaspi_program.o objdir/fpgautil_io.o objdir/tps56520.o objdir/naeusb/XMEGANVM.o objdir/naeusb/XPROGNewAE.o objdir/naeusb/XPROGTarget.o objdir/naeusb/XPROGTimeout.o objdir/naeusb/circbuffer.o objdir/tasks.o objdir/naeusb/naeusb.o objdir/naeusb/naeusb_default.o objdir/naeusb/naeusb_fpga_target.o objdir/naeusb/udc.o objdir/usart_serial.o objdir/sleepmgr.o objdir/tc.o objdir/system_sam3u.o objdir/udi_vendor_desc.o objdir/read.o objdir/interrupt_sam_nvic.o objdir/cycle_counter.o objdir/init.o objdir/uart.o objdir/syscalls.o objdir/sleep.o objdir/twi.o objdir/pio.o objdir/chipid.o objdir/exceptions.o objdir/spi.o objdir/efc.o objdir/write.o objdir/pmc.o objdir/ui_cw305.o objdir/sysclk.o objdir/startup_sam3u.o objdir/led.o objdir/udphs_device.o objdir/pio_handler.o objdir/smc.o objdir/udi_vendor.o objdir/usart.o objdir/flash_efc.o --output ChipWhisperer-CW305-SAM3U1C.elf -mthumb -Wl,--start-group -L .././hal/sam3u1c/ -lm -Wl,--end-group   -T .././hal/sam3u1c/flash.ld -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--cref -mthumb -Wl,-Map=ChipWhisperer-CW305-SAM3U1C.map,--cref   -lm  
/export/home/cmomin/tools/arm/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: objdir/naeusb/naeusb_default.o: in function `naeusb_sam_cfg_out':
/home/cmomin/tools/chipwhisperer/hardware/victims/cw305_artixtarget/fw/sam3u/CW305_SAM3U_FW/src/naeusb/naeusb_default.c:60: undefined reference to `switch_configurations'
collect2: error: ld returned 1 exit status
make: *** [.././Makefile.inc:480: ChipWhisperer-CW305-SAM3U1C.elf] Error 1
...
```
I don't have Atmel Studio installed yet. I'll try with it if no easy fix can be found.

alex-dewar, 2022-07-27:
I've confirmed the naeusb branch does work and compile, if you want to try that

cmomin, 2022-07-29:
Hi,

I tried on the naeusb branch. It compiles and the generated firmware seems to work with the aes example. Thx!

alex-dewar, 2022-08-02:
I should be merging naeusb into develop soon, so I'd just wait for that

alex-dewar, 2022-08-11:
Closing this, as the fixes have been merged into develop

closed 2022-08-11 by alex-dewar