newaetech/chipwhisperer #418: Incorrect key for plaintext XOR leakage in AES

KarthikGanesan88, 2022-09-01:
Hi, this is a question rather than an issue. I am trying [Lab4_3](https://github.com/newaetech/chipwhisperer-jupyter/blob/d8d5fc194a0d69a63a62379779c1eeab0bd66b54/courses/sca101/Lab%204_2%20-%20CPA%20on%20Firmware%20Implementation%20of%20AES%20(MAIN).ipynb) in SCA101 and it works as expected when attacking the SBOX lookup. But when I changed the leakage to:

` leak_model = cwa.leakage_models.plaintext_key_xor` 

I get 9/16 of the key bytes as having a PGE of 1 rather than 0. I thought that maybe this had to do with the XOR operation not being the biggest power consumer during the AES encrypt. So I tried modifying [`simpleserial-base.c`](https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/firmware/simpleserial-base/simpleserial-base.c) as follows:

```
uint8_t key[16] = {0};

uint8_t get_key(uint8_t* k, uint8_t len) {   
	for (int i=0; i<len; i++){
	    key[i] = k[i];
	}
	return 0x00;
}

uint8_t get_pt(uint8_t* pt, uint8_t len) {
	trigger_high();

	for (int i=0; i<len; i++){
	    pt[i] = pt[i] ^ key[i];
	}

	trigger_low();
	simpleserial_put('r', 16, pt);
	return 0x00;
}
```
In this case, I would think reading the key from memory would be the biggest power consumer and would match the `plaintext_key_xor` leakage function. But I am still getting 9/16 (the same ones) with a PGE of 1. 

Is there something I am doing wrong? Or is there some way for me to change the setup to improve the results I am getting? Thank you!

alex-dewar, 2022-09-02:
Hi,

What you're running into here is that, unlike with a non linear leakage model, the sign of the correlation actually matters here. I won't go into much detail here (can discuss further on the [forums](https://forum.newae.com/) if you'd like), but basically `key` and `~key` have the exact same leakage, but the signs are flipped

closed 2022-09-02 by alex-dewar

KarthikGanesan88, 2022-09-02:
Thank you, @alex-dewar. I'll post on the forum for help.