newaetech/chipwhisperer #419: Nano glitch too long

alex-dewar, 2022-09-13:
https://forum.newae.com/t/chipwhisperer-nano-glitch-issue/3485/4

pod2g, 2022-09-13:
Hi.

I have taken a look at https://github.com/newaetech/chipwhisperer/blob/develop/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c and this seem to be optimisable...
There is C code embedded inside  assembly...

Wouldn't it be possible to generate the assembly code when setting up the glitch offset/length? And replace the for loops with generated NOPs?

So that there's actually no instruction when scope.glitch.repeat is 1 in between the pin state changes? And a single NOP when 2, etc.

Also maybe limit scope.glitch.repeat to maximum 128 for example, I don't think there's an actual use case for this long of a glitch anyway?

pod2g, 2022-09-14:
I have commented out the `for(unsigned int i = glitch_width_cnt; i != 0; i--);` loop, basically hardcoding the glitch length to the minimum possible.

On the scope, it now lasts 40ns instead of the original 150ns.

Posting a picture on the forum for reference.

Now `SOLN_Fault 2_1B - Introduction to Voltage Glitching with CWNano` still doesn't succeed, but at least I now have a "normal count" > 0.

alex-dewar, 2022-09-14:
Yeah, this should be possible. We actually do something similar with the ChipSHOUTER, so I can take a look at that.

rlangoy, 2022-09-15:
Hi @alex-dewar on  5 Aug 2020 it seems that you sucessfully runned the voltage glitching example for [Courses_fault101_SOLN_Fault 2_1B - Introduction to Voltage Glitching with CWNano](https://github.com/newaetech/chipwhisperer-tutorials/blob/master/courses_fault101_SOLN_Fault%202_1B%20-%20Introduction%20to%20Voltage%20Glitching%20with%20CWNano-CWNANO-CWNANO.rst)  using the CWNANO firmware HW Version 0.11.0. Is there any way I could reproduce this result by using an older release  like the V5.4 ?

pod2g, 2022-09-15:
> Yeah, this should be possible. We actually do something similar with the ChipSHOUTER, so I can take a look at that.
> 
> EDIT: After to talking to Colin a bit, the code generation might not be easily achievable. You'd want to run the code from SRAM, but SRAM is much slower to run code from due to hardware optimizations targeting flash.

For reference, here is what the for-loop `for(unsigned int i = glitch_width_cnt; i != 0; i--);` translates to:

```
.text:004004D4                 LDR     R3, =glitch_width_cnt
.text:004004D6                 LDR     R3, [R3]
.text:004004D8 i = R3                                  ; unsigned int
.text:004004D8                 CBZ     R3, loop.end
.text:004004DA
.text:004004DA loop                                    ; CODE XREF: cwnano_glitch_insert+A4↓j
.text:004004DA                 SUBS    R3, #1
.text:004004DC i = R3                                  ; unsigned int
.text:004004DC                 BNE     loop
.text:004004DE loop.end                                ; CODE XREF: cwnano_glitch_insert+A0↑j
```

This could at least be optimized by moving the 2 LDRs earlier, in the initialization part.

I am trying other possibilities, maybe I can come up with a better idea.

pod2g, 2022-09-15:
Following is a patch proposal. This made `SOLN_Fault 2_1B - Introduction to Voltage Glitching with CWNano` succeeded!

I got 14 successes with these parameters:
```
gc.set_range("repeat", 1, 1)
gc.set_range("ext_offset", 1, 500)
```

See scope captures of the voltage glitch in the forum:

with `scope.glitch.repeat = 1`: ~40ns
with `scope.glitch.repeat = 2`: ~60ns
with `scope.glitch.repeat = 3`: ~80ns
with `scope.glitch.repeat = 4`: ~140ns
with `scope.glitch.repeat = 5`: ~180ns

I initially tried with the original `glitch_width_case` system but results were all over the place, with for example scope.glitch.repeat = 3 being far shorter than scope.glitch.repeat = 1...

```
diff --git a/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c b/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
index d9bac12d..b73067b0 100644
--- a/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
+++ b/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
@@ -86,8 +86,7 @@ static uint32_t glitch_offset_cnt;
 /* Configure the glitch code, must be called before calling insert */
 void cwnano_setup_glitch(unsigned int offset, unsigned int length)
 {
-	glitch_width_cnt = length / 3;
-	glitch_width_case = length % 3;
+	glitch_width_cnt = length;
 	
 	glitch_offset_cnt = offset / 3;
 	glitch_offset_case = offset % 3;
@@ -102,13 +101,16 @@ void cwnano_glitch_insert(void)
 {
 	if (glitch_width_case | glitch_width_cnt){
 		__disable_irq();
-		asm("push {r5-r6}");
+		asm("push {r4-r6}");
 		asm volatile(
 		 "mov   r5,     #0x40000000\n\t"
 		 "orr   r5, r5, #0x000e0000\n\t"
 		 "orr   r5, r5, #0x00000e00\n\t"
 		 "movs	r6,     #1\n\t"
-		 );
+		 "mov   r4,     %0\n\t"
+		 : : "r" (glitch_width_cnt-1)
+		 :
+                );
 		 
 		 
 		/* The following is very hacky, but works for now (TM). Basically:
@@ -153,37 +155,19 @@ void cwnano_glitch_insert(void)
 				asm volatile("isb");
 				break;
 		}
-		 
-		 switch(glitch_width_case){
-			 case 0:	
-				 asm volatile("isb");
-				 asm volatile("str	r6, [r5, #48]");
-				 for(unsigned int i = glitch_width_cnt; i != 0; i--);
-				 asm volatile("str	r6, [r5, #52]");
-				 asm volatile("isb");
-				 break;
-				 
-			case 1:
-				asm volatile("isb");
-				asm volatile("str	r6, [r5, #48]");
-				for(unsigned int i = glitch_width_cnt; i != 0; i--);
-				asm volatile("dsb");
-				asm volatile("str	r6, [r5, #52]");
-				asm volatile("isb");
-				break;
 
-			case 2:
-				asm volatile("isb");
-				asm volatile("str	r6, [r5, #48]");
-				for(unsigned int i = glitch_width_cnt; i != 0; i--);
-				asm volatile("str	r6, [r5, #48]");
-				asm volatile("str	r6, [r5, #52]");
-				asm volatile("isb");
-				break;
-		 }		 
-		asm("pop {r5-r6}");
+		asm volatile("isb");
+		asm volatile("str	r6, [r5, #48]");
+		asm volatile("loop0:    cbz  r4, loop0.end");
+		asm volatile("          subs r4, #1");
+		asm volatile("          bne  loop0");
+		asm volatile("loop0.end:");
+		asm volatile("str	r6, [r5, #52]");
+		asm volatile("isb");
+ 
+		asm("pop {r4-r6}");
 		__enable_irq();
 	}
 }
 
-#pragma GCC pop_options
\ No newline at end of file
+#pragma GCC pop_options
```

colinoflynn, 2022-09-15:
Very cool thanks, I'll see if @alex-dewar has any concern, but if it works it looks much more sane (no more crazy case). I do remember there being some issue with the glitch widths being "out of order", though from memory I had tuned that at one point (I wonder if indeed the compiler or similar changed, and because it was done in C code and not low-level ASM it wasn't stable).

We can apply this, but if you want to do a "real" PR then you'll get that sweet github credit ;-)

alex-dewar, 2022-09-15:
@pod2g Ended up working on this at the same time as you and ended up just doing the whole function in asm and removing the offset case as well.

I'm wondering if either the glitch transistor got swapped out at some point, or if there's a difference between batches, as I don't see any effect until I reach a repeat of 3. I can see the gate of the transistor changing, so I'm guessing the glitch is just too short to have any effect. 

@rlangoy I'm guessing this is a hardware difference, not anything to do with firmware, as the glitch code hasn't changed since the Nano was released.

pod2g, 2022-09-16:
> I'm wondering if either the glitch transistor got swapped out at some point, or if there's a difference between batches, as I don't see any effect until I reach a repeat of 3. I can see the gate of the transistor changing, so I'm guessing the glitch is just too short to have any effect.

I wonder if you did also remove the division by 3 in the `cwnano_setup_glitch` function?
Sorry to ask, you probably did, but sometimes simple issues can be overlooked.

alex-dewar, 2022-09-16:
Yeah, I removed the divide by 3. I can see the gate of the transistor going high (for ~20ns IIRC at a repeat of 1).

pod2g, 2022-09-16:
The MOSFET on my board is marked "337", I don't think it is a DMN3200U-7.

DMN3200U-7 should be marked 32Y, Y being the fabrication year AFAIK.

How about your board?

colinoflynn, 2022-09-16:
The 337 is the FDV337N, one of the subs we've had to do. I checked and the latest batch is using this too (we try to keep subs consistent once we do it if possible). The simple N-Channel MOSFET though has been a problem for past couple years, so more subs are plausible. It looks like that one is faster than the DMN3200U-7 we were originally using too, which would align with what you're seeing!

pod2g, 2022-09-17:
Hi Colin, thanks a lot for the precisions on the substitutions!

According to the FDV337N documentation:

Turn-On Delay Time: 10ns
Turn-On Rise Time: 18ns

That's quite a big improvement over DMN3200U-7 !! and can totally explain the difference in behaviour. Now the original code makes more sense, the 2 loads before the loop probably did not make a difference with DMN3200U-7, or rather, they would allow it to be ready to work.

The fact that @alex-dewar 's variant does not have the time to glitch at repeat=1 is probably not big of deal, considering the original Jupyter notebook does a loop from 1 to 3 originally. You could maybe bump it to 5, just to be sure.

rlangoy, 2022-09-17:
I did a simple simulation of the VDD glitching using LT-Spice to compare the difference using DMN3200U vs the FDV337N
(I am not sure about the quality of the DMN3200U  model...)
But this was my results..
#### Result of singe 60ns Pulse
[image]

### Result of multiple 60ns Pulse with 50% dutycycle
[image]

The models and simulations could be found at : https://github.com/rlangoy/cw_nano_glitch_sim

Hopes this input could helpful

rlangoy, 2022-09-22:
Hi @alex-dewar unfortunately the new firmware did not work for me.
The target on my cw-nano is a stm32f04 2f6p6 and not the original stm32f03xxx.

pod2g, 2022-09-22:
Hi @alex-dewar,

I tried my best as well, no luck with your latest code. I never get a success.
However, if I revert to my loop proposal instead, it works.

Here is an in between proposal that functions as well:

```
diff --git a/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c b/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
index 7050128c..c468793b 100644
--- a/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
+++ b/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
@@ -108,8 +108,7 @@ void cwnano_glitch_insert(void)
                                "movs   r6,     #1\n\t"
 
                                "ldr r3, %[offset_cnt]\n\t"
-                               "ldr r4, %[width_cnt]\n\t"
-
+                               "ldr r4, %[width_cnt-1]\n\t"
                                "isb\n\t"
                        "OFFLOOP:\n\t"
                                "subs r3, #1\n\t"
@@ -117,9 +116,11 @@ void cwnano_glitch_insert(void)
                                "isb\n\t"
 
                                "str r6, [r5, #48]\n\t" //gpio high now
+                               "cbz  r4, WID0LOOP.end\n\t"
                        "WID0LOOP:\n\t"
-                               "subs r4, #0x01\n\t" //1 is the minimum for width_cnt
+                               "subs r4, #1\n\t"
                                "bne WID0LOOP\n\t"
+                       "WID0LOOP.end:\n\t"
                                "str r6, [r5, #52]\n\t"
                                "isb\n\t"
 
@@ -228,4 +229,4 @@ void cwnano_glitch_insert(void)
 }
 
 #pragma GCC pop_options
-#endif
\ No newline at end of file
+#endif
```

@rlangoy: I also have the STM32F042F6P6.

Note: I used these parameters for testing:
```
gc.set_range("repeat", 1, 2)
gc.set_range("ext_offset", 1, 500)
````

@colinoflynn, @alex-dewar: with all the testing and time lost on this, I hope you guys will send us a free kit ;-) j/k

~pod

alex-dewar, 2022-09-22:
Interesting. I'd guess our loops differ by a clock cycle and that's making the difference. Out of curiosity, do you see successes with both a width of 1 and 2 with your code? I'm guessing my original loop with width 1 is ~2 cycles and both width 1 and 2 for you are ~3 cycles.

You can make a pull request with the changes if you'd like, or I can patch your changes in. 

Thanks for your help with this. We do generally like to send some thank you's out in situations like these, but parts shortage are still rough, so I can't guarantee anything.

pod2g, 2022-09-23:
Here are some measurements.
I used GPIO3, connected to a 2.7K resistor to ground. Oscilloscope in persistence mode to see the jittered traces.

Current version on GitHub:
- width=1: traces of 33ns, 49ns
- width=2: traces of 58ns, 66ns, 74ns
- width=3: traces of 83ns, 91ns, 99ns
- width=4: traces of 108ns, 107ns, 124ns

Generated asm version:
- width=1: traces of 41ns, 49ns, 58ns
- width=2: traces of 41ns, 49ns, 58ns
- width=3: traces of 58ns, 66ns
- width=4: traces of 58ns, 66ns
- width=5: traces of 83ns, 91ns, 99ns
- width=6: traces of 83ns, 91ns, 99ns
- width=7: traces of 108ns, 107ns, 124ns

All this doesn't make a lot of sense. I don't understand why the current version doesn't give any success.
Maybe the fact that it misses the 41ns glitch?

pod2g, 2022-09-25:
Just to let you know, I am currently trying to implement a PWM solution. It may work because whoever designed this board had a great idea using PIOA0 for the glitch output, which can output PWMH0 in PERIPHERAL A mode.

I have tested that it's able to generate a 1 clock cycle glitch.

Now the difficulty is doing this in the existing code, and having a solution for both width and offset.

Hopefully it's going to work out.

pod2g, 2022-09-27:
8.33ns resolution glitches with the PWM module:

```
diff --git a/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c b/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
index 7050128c..be9f0f92 100644
--- a/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
+++ b/hardware/capture/chipwhisperer-nano/firmware/cwnano-firmware/src/cwnano_glitch.c
@@ -45,21 +45,9 @@ void pin_trigglitch_handler(const uint32_t id, const uint32_t mask);
 
 static volatile unsigned int glitch_enabled = 0;
 
-/* Handler for all PIOA events */
-void pin_trigglitch_handler(const uint32_t id, const uint32_t mask)
-{
-	if ((id == ID_PIOA) && (mask == PIN_TARGET_GPIO4_MSK)){
-		
-		/* Disable interrupt now */
-		pio_disable_interrupt(PIOA, PIN_TARGET_GPIO4_MSK);
-		
-		if(glitch_width && glitch_enabled){
-			cwnano_glitch_insert();
-			glitch_enabled = 0;
-		}		
-	}
-}
-
+static uint32_t glitch_width_cnt = 0;
+static uint32_t glitch_offset_cnt = 0;
+static uint32_t glitch_wait_cnt = 0;
 
 void cwnano_glitch_enable(void)
 {
@@ -69,163 +57,109 @@ void cwnano_glitch_enable(void)
 	pio_handler_set(PIOA, ID_PIOA, PIN_TARGET_GPIO4_MSK, PIO_IT_RISE_EDGE, pin_trigglitch_handler);
 	pio_enable_interrupt(PIOA, PIN_TARGET_GPIO4_MSK);
 	glitch_enabled = 1;
+
+	PWM->PWM_DIS = PWM_DIS_CHID0;                                                // channel 0 must be disabled before cwnano_glitch_insert is invoked
 }
 
+#define PWM_GPIO0 IOPORT_CREATE_PIN(PIOA, 0)
+
 /* Init the glitch pin (drive low) */
 void cwnano_glitch_init(void)
 {
-	gpio_configure_pin(PIN_GLITCH_IDX, PIO_OUTPUT_0 | PIO_DEFAULT);
-}
+	PMC->PMC_PCER0               = (1<<ID_PWM);                                  // clock on the PWM module
+	PIOA->PIO_PDR                = PIO_PDR_P0;                                   // PA0 pin is not PIO anymore
+	PIOA->PIO_ABCDSR[1]         &= ~PIO_ABCDSR_P0;                               // PA0 pin should use peripheral A mode
 
-static uint32_t glitch_width_case;
-static uint32_t glitch_width_cnt;
-
-static uint32_t glitch_offset_case;
-static uint32_t glitch_offset_cnt;
+}
 
 /* Configure the glitch code, must be called before calling insert */
 void cwnano_setup_glitch(unsigned int offset, unsigned int length)
 {
-	glitch_width_cnt = length;
-	
-	glitch_offset_cnt = offset;
+	uint32_t glitch_prd;
+	uint32_t glitch_duty;
+
+	if ((length > 0 && length != glitch_width_cnt) || (offset > 0 && offset != glitch_offset_cnt)) {
+		glitch_width_cnt             = length;
+		glitch_offset_cnt            = offset;
[... 186 lines cut ...]
+		"subs  r0,     #1             \n\t"
+		"bne   loop.wait.%=           \n\t"
+		:
+		: [wait_cnt] "m" (glitch_wait_cnt)
+		: "r0", "memory"
+	);
+
+#if 0
+	// debug: output a blip after the glitch to fine tune the wait time with a scope
+	PWM->PWM_OOV                 = PWM_OOV_OOVH0;    // override: value 1
+	PWM->PWM_OS                  = PWM_OS_OSH0;      // override: output 1, not taking care of the pwm output
+	PWM->PWM_OOV                 = ~PWM_OOV_OOVH0;   // override: value 0
+#else
+	PWM->PWM_OS                  = PWM_OS_OSH0;      // override: output 0, not taking care of the pwm output
+#endif
+
+	asm volatile(
+		"cpsie i                      \n\t"
+	);
 }
 
 #pragma GCC pop_options
-#endif
\ No newline at end of file
```

pod2g, 2022-09-27:
With the PWM, we can also change the frequency to be nearly anything between 0 and 120MHz so that the glitch length doesn't necessarily need to be a multiple of 8.33ns.

Reason I am saying this is that the PWM version seems great from the scope perspective. It's incredible to see those 8.33ns perfect glitches. However it does not lead any success on my board, exactly like the current version on GitHub. The issue is probably more related to the synchronous nature of this?

@alex-dewar : what do you think?

rlangoy, 2022-09-29:
@pod2g  I realy appreciate all your hard work :) , but unfortunately it did not work for me..
The problem seeme to me that when the glitch drops below 600mV the circuit vould reset.
Adding a resistor to privide some bias helped me..
[image]

By desoldering the Sj6 and adding a 5Ohm resistor between GLITCH and Measure i did manage to get som sucesses
(The R12 in the schematic was not 27Ohm , but 30Ohm on my card..)
[image]
This is scope output:
[image]

alex-dewar, 2022-09-29:
Based on the results you guys have seen, I think it's best if we either do something similar to the original loop, or the generated asm version. Regarding the latter, do you see any difference if you do a different single cycle operation, like an add? I'm not sure about others, but I believe `nop` isn't guaranteed to take any time (i.e. the CPU can skip them). 

I'd agree that the PWM version definitely opens up some interesting glitch opportunities, so it'd be nice to at least keep it as an alternative. Maybe we could keep it `#ifdef`'d out/incomplete and mention it in the hardware docs?

pod2g, 2022-09-29:
@rlangoy : absolutely great mod. Thanks for sharing! I think this adds another perspective to the research.

This as well as the 60MHz PWM test seems to tell that the length and resolution of the glitch are not the only parameters to achieve success.

What I find good about the PWM is that we can artificially control the voltage of the glitch through software.

What would be nice would be to have a PoC with all parameters (period, duty, length) accessible to the user from Python and test what is possible to achieve with that. Unsure I'll have the time to work on this through.

@alex-dewar: good point about the nop. I can try other instructions and see the result for the generated assembly version.

alex-dewar, 2022-09-29:
If you'd like, I can do the USB/Python code up for the PWM glitch part.

pod2g, 2022-09-29:
That would be great! Thanks for proposing.

alex-dewar, 2022-09-29:
Also quick question, do you want me to test out the add version of the generated assembly as well? I should be able to get one of the nano's with the new transistor to verify that the glitch works.

pod2g, 2022-09-29:
Sure! I'll report though if I get a chance to try. Thanks again!

alex-dewar, 2022-10-11:
Quick update, still working on this. Turns out a precise offset is also very important here, so I ended up implementing a faster version of the original, with multiple offset cases. This seems to work for me for both the new and old transistor.

I did find something interesting: writing to the IO pin (one or more times) then doing a nop as a delay results in almost no jitter on the glitch width. Any other instructions (two nops in a row, any branching, etc.) results in the same jitter as before. Also I've found that increasing the target clock speed like we do with the Lite voltage glitching labs results in many more successful glitches.

pod2g, 2022-10-12:
Hi @alex-dewar.

Is the version you're talking about already on the repo?

I'd love to test this out!

No rush though, I'm out to Hexacon for the next few days, getting back home on Sunday.

There's also the delay to glitch of approx 4 microseconds that would be nice to address (if possible?).

Do you think there's room for improvement in the interrupt handler?

Maybe a new ticket for this other subject?

Awesome news, and Thanks!

~pod

alex-dewar, 2022-10-14:
Just pushed to a new branch: https://github.com/newaetech/chipwhisperer/tree/nano_glitch. Still pretty WIP, but should compile and work

Re the delay on the interrupt, it might be possible to improve performance here, but it's hard to say. Arm gives 12 cycles of interrupt latency, but that's based on a 0 wait state flash model. It's also hard to say how much of the latency comes from unnecessary/unoptimized operations. It might be interesting to look at in the future, but there's currently a lot of stuff for me that's higher priority.

alex-dewar, 2023-10-13:
The above branch got merged in a while back, so closing this issue.

closed 2023-10-13 by alex-dewar

grasshoppper4, 2024-05-01:
Today I just updated the firmware and this issue occurs.